Inspire-Poli-USP/Inspire-OpenLung #11: Fritzing - Desenhar ligação eletrônica arduino|controle

HPparanhos, 2020-04-01:
Pessoal, me mandem umas fotos de referência da ligação de vocês no whatsapp.

Amanhã vou sentar para realizar os desenhos das nossas ligações e aproveito para fazer a versão de vocês. Precisamos documentar toda parte de ligação do arduino para que pessoal possa replicar os testes em outros lugares.

Vamo que vamo!

atodobom, 2020-04-01:
Olá paranhos. Como pretende fazer a interface?
Estou pensando em uma IHM com tablet, conectado via Bluetooth.
Por hora enquanto discutimos a possibilidade de colocar um oxímetro, teremos apenas a frequência respiratória. O Arduino pode enviar via SPP em BT, usando um módulo HC05.
No tablet penso em ter o nr do quarto, leito, nome do paciente, e gerar alertas em caso de desconexão com o aparelho, ou falhas reportadas.

Eventualmente podemos fazer mais com isso. Apenas agregando mais no APK.
Conversando do Dr. Sidomar Cardoso, ele me passou algumas colocações.
Ele já foi médico militar no serviço amazônico,  e trabalhou muito em locais remotos e sem recursos. Hoje atende pela rede pública e privada em Rondônia.
Ele me passou que seria muito interessante agregar um oxímetro ao produto. Monitores multiparâmetros não são comuns em todos os locais e serão raros nessa demanda.
Nesta semana perderam um paciente por conta disso. O ventilador estava operando, mas ninguém observou a queda de saturação, e quando viram o ventilador estava forçando ar num paciente em óbito.

Outra coisa ele falou que podemos agregar funcionalidades para organizar um pouco a coisa. Ele acredita que teremos condições difíceis em campo. E se tiver dados do paciente seria bom.
Se houver um tablet por aparelho, eles podem estar em rede. E existir um tablet ou qualquer celular no posto de enfermagem, com um apk de supervisão. Ele pode dar alerta na falha de algum aparelho ou perda de comunicação.
O tablet do posto pode lançar posologias que podem aparecer nos que estão junto ao paciente. Gerando log de tudo. Enfim... discussão aberta. Vamos considerar opiniões de especialistas para nos conduzir nisso.
Estou aqui com Multilaser M7S-GO baixo custo. Android 8 e utilizei para outros projetos.

Estou iniciando um projeto em plataforma AI2 que ficará em aberto para comunidade.

HPparanhos, 2020-04-01:
https://github.com/HPparanhos/CITI-OpenLung/tree/master/Images/Video_test

Parte de controle já foi feita usando o arduino mesmo, ele fará o controle de telemetria e o CLP fará o controle de motor e válvulas. Os dois terão comunicação entre si.

Estamos adaptando novos sensores e brincando um pouco com Blender, em breve teremos alguns vídeos para publicação. 

Abraço!

HPparanhos, 2020-04-01:
Sobre um produto com Wi-fi ou bluetooth, ele deve ser submetido a duas provas de inmetro especifico, para um produto de mercado no caso. Cada teste de validação gera em torno de 20K Reais, e pode ser facilmente reprovado. 

Não sei ao certo como seria neste caso junto a ANVISA.

 Mas creio que o mais simples é adequado para situação que estamos prevendo.

atodobom, 2020-04-01:
O usuário, médico ou enfermeiro, vai precisar interagir com o CLP? Isso seria complicado para certificar.
Podemos fazer a IHM simplificada e através dela realizar os ajustes necessários no CLP?
Por exemplo, o perfil do paciente, alterando o volume de ar fornecido por respiração e a frequência respiratória.

HPparanhos, 2020-04-01:
No vídeo que mandei a cima tem a interface. Toda via potenciômetro e tela no arduino. CLP apenas controla os motores.

AlvaroAguiar, 2020-04-15:
> https://github.com/HPparanhos/CITI-OpenLung/tree/master/Images/Video_test
>
> Parte de controle já foi feita usando o arduino mesmo, ele fará o controle de telemetria e o CLP fará o controle de motor e [...]

O link não leva a uma página válida.

breno-helf, 2020-04-15:
Oi @AlvaroAguiar acho que tiraram o link para o Vídeo test, acredito que foi por questões regulatórias. @OttoHeringer pode confirmar?

closed 2020-04-26 by breno-helf