Inspire-Poli-USP/Inspire-OpenLung #12: Oxímetro

atodobom, 2020-04-01:
Seguindo opiniões profissionais, já houve perda de paciente por falta de oxímetro.
O produto aqui tem como objetivo final oxigenar o paciente. Monitores multiparâmetros serão raros na atual demanda.
Estive estudando alguns dispositivos. Mas se a gente tentar usar esses simples apresentam alguns problemas, como utilizar pilhas que não duram mais que 12 horas, e nem sempre se mantém ligados. Quando conectam via bluetooth nem sempre temos o protocolo aberto e problemas de certificação.
Fui atrás de sondas certificadas, encontrei algumas, mas são discretas e basicamente tratam-se de sensores óticos, led, e fototansistor e precisam ter os dados tratados com circuitos conformadores de sinal, e algorítimos complexos para se obter a saturação e pulsação.
Encontrei uma placa muito pequena interessante. Ela é composta dos sensores e já um ARM cortex M0 e saída I2C já com os dados tratados. O problema dela seria construir o suporte mecânico para acoplar ao dedo do paciente, tendo algumas versões conforme idade. E saber o que isso impacta na certificação final do produto. No site apresentam programa de teste para Arduino. É simples, ele envia algumas vezes por segundo os dados de saturação, pulso e um valor que apresenta a qualidade da medição, para caso esteja saindo do dedo gerar um alerta.

VitorFrost, 2020-04-02:
Muito interessante, mas isso não impediria o uso de desfibriladores?

akapmarak, 2020-04-02:
É uma demanda pertinente.
Porém o valor produto final não seria desproporcional?

Um oxímetro está uns 70 reais, já com display. E daria pra adaptar uma fonte de energia.

Diferencial seria uma telemetria com módulos de maior alcance, pra controlar toda uma enfermaria.

atodobom, 2020-04-02:
Problema é um oxímetro desse de R$ 70 com pilha que não dura muito. Sem conexão ele não gera alerta, ao menos não seja chamativo para o ambiente que existirá, e se conectar por bluetooth não sei se acharemos com protocolo aberto, além das possíveis dificuldades quanto a aquisição de determinados modelos etc.
Podemos pensar quando ao material que coloquei, o módulo microcontrolado que possui já o tratamento do sinal e entrega as medidas prontas, ou usar probes hospitalares certificadas e tratar o sinal.
No mercado local, e fabricado aqui tempos as probes que são "compatíveis" com equipamentos.
Trata-se de uma probe para colocar no dedo, contendo led Red e iRed, e fotosensor.

Encontrei o material para tratar o sinal em https://www.bbc.com/portuguese/internacional-40734577

Pessoalmente eu gosto de usar um módulo com ARM que tem várias entradas ADC, seriais, e já tem BLE com antena e tudo. O módulo tem menos de 1 cm², com bom poder computacional.
Seria uma opção, mas me questiono o Time to Market disso.

VitorFrost, 2020-04-02:
Antonio, acredito que o link esteja incorreto.
Verifique se era esta referência: https://www.nxp.com/docs/en/application-note/AN4327.pdf

atodobom, 2020-04-02:
Hauhaua... 
Não acredito que esteja incorreto... eu tenho CERTEZA! 
Sim.. essa AN mesmo. Muita coisa aberta no browser dá nisso! KKKK
Obrigado! Já até editei o comment. Mas assim.
Entendam que são ideias soltas estilo brainstorm, que podem eventualmente ter algum resultado

FlavioAlegretti, 2020-04-05:
Olá pessoal, em alguma parte do projeto pensam em utilizar conversores para acionamento do motor ou estão acionando direto das pilhas? Digo isso pois lendo a discussão percebi a preocupação com as pilhas e talvez um conversor (seria ligado na tomada 127/220 e convertido para 12V, 9V ou 5V) forneceria com pouca utilização de espaço a energia necessária (um conversor flyback estilo utilizado em carregadores de celular), único inconveniente seria ter que ligar na tomada.

closed 2020-04-26 by breno-helf