Inspire-Poli-USP/Inspire-OpenLung #14: ANVISA

m-rodrigues-souza, 2020-04-01:
O que já temos de informação sobre a certificação do produto? Em qual classe ele se enquadra, quais as normas aplicáveis e qual o processo para certificar?

emersonmoretto, 2020-04-01:
Opa, ainda nao temos, estamos validando a capacidade dele de fluxo,
pressao no pulmao artificial da usp e devemos partir pra validacao em
animais e regulaçao a partir de segunda

valeu

*Emerson Moretto*
emerson@moretto.eng.br
11 99484-0571

atodobom, 2020-04-01:
Ao menos as NBR ISO 62366 e 60601-1-6 parecem que precisam ser seguidas.
Aí tem todas aquelas RDC que indicam outras normas.

OttoHeringer, 2020-04-03:
Acabei de receber uma info e docs sobre regulamentação. Vou criar um documento explicativo e fazer upload das normas que temos aqui.
Obrigado por ter criado esse Issue, @atodobom !

OttoHeringer, 2020-04-06:
Um ótimo resumo do tamanho do trabalho que seria fazer isso "normalmente" é esse:

[image]

OttoHeringer, 2020-04-06:
Outra informação que recebi em grupos de whatsapp com pessoas envolvidas com esse trabalho regulatório tem a seguinte listagem de normas (que é basicamente um dos quadradinhos do fluxograma acima):

1. ABNT NBR IEC 60601-1:2010 - Equipamento eletromédico - Parte 1: Requisitos gerais para segurança básica e desempenho essencial

2. ABNT NBR IEC 60601-1:2010  Emenda 1:2016 - Equipamento eletromédico - Parte 1: Requisitos gerais para segurança básica e desempenho essencial
 
3. ABNT NBR IEC 60601-1-2:2017 - Equipamento eletromédico - Parte 1-2: Requisitos gerais para segurança básica e desempenho essencial - Norma Colateral: Perturbações eletromagnéticas - Requisitos e ensaios

4. ABNT NBR IEC 60601-1-6:2011 Versão Corrigida 2013 - Equipamento eletromédico - Parte 1-6: Requisitos gerais para segurança básica e desempenho essencial - Norma colateral: Usabilidade
 
5. ABNT NBR IEC 60601-1-8:2014 Versão Corrigida 2015 - Equipamento eletromédico - Parte 1-8: Requisitos gerais para segurança básica e desempenho essencial - Norma colateral: Requisitos gerais, ensaios e diretrizes para sistemas de alarme em equipamentos eletromédicos e sistemas eletromédicos

6. ABNT NBR IEC 60601-1-10:2010  Versão Corrigida 2013 - Equipamento eletromédico - Parte 1-10: Requisitos gerais para segurança básica e desempenho essencial - Norma colateral: Requisitos para o desenvolvimento de controladores fisiológicos em malha fechada
 
7. ABNT NBR IEC 60601-1-10:2010 Emenda 1:2017 - Equipamento eletromédico - Parte 1-10:
Requisitos gerais para segurança básica e desempenho essencial — Norma colateral: Requisitos para o desenvolvimento de controladores fisiológicos em malha fechada

8 - ABNT NBR ISO 80601-2-12:2014 - Equipamento eletromédico - Parte 2-12: Requisitos particulares para a segurança básica e o desempenho essencial de ventiladores para cuidados críticos

9 - ABNT NBR ISO 14971:2009 - Produtos para a saúde — Aplicação de gerenciamento 
de risco a produtos para a saúde

10 - ABNT NBR ISO TR 24971:2014 - Produtos para a saúde - Orientações para a aplicação da ABNT NBR ISO 14971

11 - ABNT NBR IEC 62366: 2016 - Produtos para a saúde — Aplicação da engenharia de usabilidade a produtos para a saúde

OttoHeringer, 2020-04-06:
Há pessoas comentando sobre a criação de alguma portaria "flexibilizante", como aconteceu nos (lamentáveis) casos da cloroquina e fosfoetanolamina, o que pode ser tanto uma coisa boa como ruim no nosso caso dos respiradores, na minha opinião.

Outros comentários estão na linha de criar um "fast track", que eu entendo como sendo alguma maneira de "encurtar caminho" naquele processo todo do fluxograma [no comentário anterior deste issue](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/issues/14#issuecomment-609779681).

buenow, 2020-04-06:
Pessoal, o fast track diz respeito a produto já aprovado no exterior, pela FDA, Ag Européia, Japonesa ou Chinesa. Neste caso a aprovação será imediata se o projeto tiver alguma dessas aprovações externas. Por outro lado, o prazo da ANVISA foi reduzido para apreciação desse tipo de produto, em caso de projeto local. Creio ainda que sendo a USP esse processo tenha alguma prioridade junto à ANVISA. A imprensa informa que já existe um contato estabelecido para tal finalidade entre a USP e a ANVISA. Isso procede? Como podemos ajudar?

OttoHeringer, 2020-04-06:
Entendi. Pensei que fast track pudesse significar mais coisas além disso que você mencionou.

Sim, já existe uma conversa com alguns profs e a ANVISA. Apesar de possivelmente o tempo de apreciação ser menor, ainda é uma lista enorme de testes e exigências como as que mostrei antes.

O problema é que além dos testes do protótipo, é preciso pedir autorização para o LOCAL de fabricação (o começo do fluxograma que mostrei antes).

ArturGasparetto, 2020-04-07:
Olá pessoal, de que forma podemos contribuir com esta parte regulatória?
Eu consegui algumas das normas citadas acima pelo site da ABNT e gostaria de saber se tem algum local já definido de onde compartilhar estas normas.

xultz, 2020-04-07:
Recebi um doc contendo uma lista de normas que a ABNT está disponibilizando gratuitamente, quase metade delas se referem a ventiladores (de alguma forma), o arquivo pode ser baixado aqui:
https://gofile.io/?c=RAMekj
(Desculpe ser um arquivo .doc, mas foi o que recebi, se alguém tiver dificuldade eu converto para um formato mais acessível)

guiriole, 2020-04-07:
Boa tarde pessoal,
Sou da Riole (www.riole.com.br). Vamos começar a contribuir no desenvolvimento e possível fabricação. Eu vou tratar das questões burocráticas e estou tentando me inteirar.
Mas pelo que estudei até aqui a fabricação esta dispensada de qualquer autorização conforme a RDC nº356 de 23/03/2020.(http://portal.anvisa.gov.br/documents/10181/5809525/RDC_356_2020_.pdf/0655c7ae-8c47-4be9-bf0d-4c7b8df03e4e). 
Na mesma RDC já se estabelece alguns requisitos do produto (art. 7º).

ArturGasparetto, 2020-04-07:
Só para entender...a RDC 356 trata de EPIs e também de válvulas, circuitos e conexões respiratórias que serão usadas no ventilador?
Se for isso, então acho que a RDC 356 tem que ser usada em conjunto com a RDC 349 19/03/2020, certo?

guimoro, 2020-04-07:
> Só para entender...a RDC 356 trata de EPIs e também de válvulas, circuitos e conexões respiratórias que serão usadas no ventilador?
> Se for isso, então acho que a RDC 356 tem que ser usada em conjunto com a RDC 349 19/03/2020, certo?

Me parece que sim.

ArturGasparetto, 2020-04-07:
É, foi meu entendimento também.
Como forma de ajudar, vou pesquisar e estudar para tentar entender a legislação, normas, etc que envolvem o processo regulatório e de cadastramento, avaliação de conformidade e liberação para uso do ventilador.
Me disponho aqui a ficar em contato com quem mais estiver ajudando nesta área que é bem ampla e cheia de detalhes.

ArturGasparetto, 2020-04-07:
Estou usando como base para esse estudo de todo processo regulatório, o documento "Manual para regularização de equipamentos médicos na Anvisa" emitido em Abril de 2020.
[Manual para regularização de equipamentos médicos na Anvisa.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4447091/Manual.para.regularizacao.de.equipamentos.medicos.na.Anvisa.pdf)

OttoHeringer, 2020-04-07:
@ArturGasparetto obrigado por se disponibilizar em compartilhar as normas. Elas não são pagas? Se depositarmos esses pdfs por aqui não poderia "dar ruim" pra gente? Mas se tiver um link com elas, acho que não teria problema.
@guiriole, eu tinha lido essa portaria mas só caiu a ficha disso com o seu comentário, rs. Tem razão, as indústrias estão dispensadas.

Pensei numa maneira de organizar a ajuda que acho que ia ajudar MUITO: ler as normas e "filtrar" delas todos os requisitos técnicos necessários. Até agora ninguém de fato teve "coragem" de enfrentar esse "calhamaço" de normas.
Podem ir postando direto aqui mesmo, posts soltos. Depois eu junto bonitinho - ou algum de vcs pode juntar tbm e mandar um pull-request, se fizerem questão.

Aí transformaríamos essa discussão e os pontos numa página da wiki desse repositório, onde outros poderão se interar dessa questão rapidamente.

Eu queria já ter feito isso, mas me faltou tempo. Podem ajudar nisso!?

ArturGasparetto, 2020-04-07:
Oi @OttoHeringer , as normas que obtive foram apenas as que estavam disponíveis gratuitamente pelo site abntcatalogo.com.br . Contudo, considerando o que você disse sobre a possibilidade de dar algum problema, posso sim tentar disponibilizar os links aqui de forma a facilitar que cada um baixe esses documentos. Mas lembrando que algumas normas não consegui baixar pois eram pagas mesmo.
Se for de concordância e acharem necessário também, posso colocar alguns outros documentos normativos como INs, RDCs, etc que considero relevante no processo de credenciamento, certificação e liberação de uso do dispositivo, de forma a tentar centralizar toda documentação aqui.
Eu tenho mais 2 colegas de trabalho que estão estudando essas normas também, tentando entender tudo que envolve o funcionamento deste tipo de dispositivo e mapear os requisitos técnicos baseando-se nessa pilha de normas.

ArturGasparetto, 2020-04-08:
Adicionei os links do site do catálogo da ABNT para as normas que foram listadas pelo @OttoHeringer . Todas essas normas encontram-se como gratuitas no momento.

1. ABNT NBR IEC 60601-1:2010 - Equipamento eletromédico - Parte 1: Requisitos gerais para segurança básica e desempenho essencial [https://www.abntcatalogo.com.br/norma.aspx?ID=80437](url)

2. ABNT NBR IEC 60601-1:2010  Emenda 1:2016 - Equipamento eletromédico - Parte 1: Requisitos gerais para segurança básica e desempenho essencial [https://www.abntcatalogo.com.br/norma.aspx?ID=355495](url)

3. ABNT NBR IEC 60601-1-2:2017 - Equipamento eletromédico - Parte 1-2: Requisitos gerais para segurança básica e desempenho essencial - Norma Colateral: Perturbações eletromagnéticas - Requisitos e ensaios [https://www.abntcatalogo.com.br/norma.aspx?ID=383029](url)

4. ABNT NBR IEC 60601-1-6:2011 Versão Corrigida 2013 - Equipamento eletromédico - Parte 1-6: Requisitos gerais para segurança básica e desempenho essencial - Norma colateral: Usabilidade [https://www.abntcatalogo.com.br/norma.aspx?ID=307130](url)

5. ABNT NBR IEC 60601-1-8:2014 Versão Corrigida 2015 - Equipamento eletromédico - Parte 1-8: Requisitos gerais para segurança básica e desempenho essencial - Norma colateral: Requisitos gerais, ensaios e diretrizes para sistemas de alarme em equipamentos eletromédicos e sistemas eletromédicos [https://www.abntcatalogo.com.br/norma.aspx?ID=348135](url)

6. ABNT NBR IEC 60601-1-10:2010  Versão Corrigida 2013 - Equipamento eletromédico - Parte 1-10: Requisitos gerais para segurança básica e desempenho essencial - Norma colateral: Requisitos para o desenvolvimento de controladores fisiológicos em malha fechada [https://www.abntcatalogo.com.br/norma.aspx?ID=307136](url)

7. ABNT NBR IEC 60601-1-10:2010 Emenda 1:2017 - Equipamento eletromédico - Parte 1-10: Requisitos gerais para segurança básica e desempenho essencial — Norma colateral: Requisitos para o desenvolvimento de controladores fisiológicos em malha fechada [https://www.abntcatalogo.com.br/norma.aspx?ID=371257](url)

8. ABNT NBR ISO 80601-2-12:2014 - Equipamento eletromédico - Parte 2-12: Requisitos particulares para a segurança básica e o desempenho essencial de ventiladores para cuidados críticos [https://www.abntcatalogo.com.br/norma.aspx?ID=326196](url)

Obs.: _As normas técnicas listadas acima foram aprovadas pela Instrução Normativa do Ministério da Saúde nº 49 de 27 de Novembro de 2019 ([IN 49 Ministério da Saúde-NOVEMBRO DE 2019.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4447684/IN.49.Ministerio.da.Saude-NOVEMBRO.DE.2019.pdf)), cujos parâmetros devem ser adotados para a certificação de conformidade, no âmbito do Sistema Brasileiro de Avaliação da Conformidade (SBAC), dos equipamentos sob regime de Vigilância Sanitária._

9. ABNT NBR ISO 14971:2009 - Produtos para a saúde — Aplicação de gerenciamento de risco a produtos para a saúde [https://www.abntcatalogo.com.br/norma.aspx?ID=55540](url)

10. ABNT NBR ISO TR 24971:2014 - Produtos para a saúde - Orientações para a aplicação da ABNT NBR ISO 14971 [https://www.abntcatalogo.com.br/norma.aspx?ID=325424](url)

11. ABNT NBR IEC 62366: 2016 - Produtos para a saúde — Aplicação da engenharia de usabilidade a produtos para a saúde [https://www.abntcatalogo.com.br/norma.aspx?ID=354941](url)

guiriole, 2020-04-08:
Obrigado ArturGasparetto,

Sobre a permissão da ANVISA para fabricação, o manual que você disponibilizou é bem claro.
É preciso obter Certificação de Boas Práticas de Fabricação emitida pela Anvisa. **Quem possui ISO 9001 esta dispensando pelo RDC 349**. Nós possuímos a ISO 9000, então é preciso obter. Por isso:

**PASSO 1:** Cadastrar a empresa (https://www9.anvisa.gov.br/recadastramento/Login.asp?SID=650422935)

Estou já com dificuldades. A definição do porte da empresa é obrigatória, mas não está sendo possível selecionar o porte.
Amanhã tentarei entrar em contato na Central de Atendimento Anvisa - 0800 642 9782 para ver uma solução, porque me parece um bug do sistema.

Sobre a pergunta inicial, a Classe de Risco é III, segundo o RDC nº 185.

buenow, 2020-04-08:
Pessoal, a RDC_356_2020 não libera a fabricação das regulamentações. Acho que este entendimento é equivocado. Me corrijam se estiver errado, mas o artigo 9º é o único que menciona os ventiladores pulmonares e diz - Art. 9° Fica permitida a **aquisição** de equipamentos de proteção individual,
**ventiladores pulmonares**, circuitos, conexões e válvulas respiratórios, monitores
paramétricos e outros dispositivos médicos, essenciais para o combate à COVID-19, novos
e não regularizados pela Anvisa, desde que regularizados e comercializados em jurisdição
membro do International Medical Device Regulators Forum (IMDRF), por órgãos e
entidades públicas e privadas, bem como serviços de saúde, quando não disponíveis para o
comércio dispositivos semelhantes regularizados na Anvisa.

Vejam, trata-se de aquisição dos aparelhos em outra jurisdição. Não fala em fabricação.

ArturGasparetto, 2020-04-08:
@buenow Acho que você está correto, inclusive considerando também o Artigo 2º que diz:
"_Art. 2° A fabricação e importação de máscaras cirúrgicas, respiradores particulados N95, PFF2 ou equivalentes, óculos de proteção, protetores faciais (face shield), vestimentas hospitalares descartáveis (aventais/capotes impermeáveis e não impermeáveis), gorros e propés, válvulas, circuitos e conexões respiratórias para uso em serviços de saúde ficam excepcional e temporariamente dispensadas de Autorização de Funcionamento de Empresa, da notificação à Anvisa, bem como de outras autorizações sanitárias._"
O artigo, por citar certos equipamentos específicos, dá a entender que apenas esses tipos de equipamentos estariam temporariamente dispensados de autorização de funcionamento e outras autorizações para sua fabricação e importação.

buenow, 2020-04-08:
"Infelizmente" Artur, é isso aí... Meu nome é Wilson Bueno - buenow...  Obrigado pelo retorno

ArturGasparetto, 2020-04-08:
Gostaria de disponibilizar aqui um documento importante para a etapa de registro de produto na ANVISA.
A RDC 185/2001 da ANVISA ([RDC 185-OUTUBRO 2001.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4450659/RDC.185-OUTUBRO.2001.pdf)) que trata do regulamento para registro de produtos médicos.
Eu achei um documento de orientações da RDC 185/2001, que talvez seja útil nesta etapa do processo. ([RDC 185-OUTUBRO 2001-MANUAL DO USUÁRIO.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4450694/RDC.185-OUTUBRO.2001-MANUAL.DO.USUARIO.pdf))

buenow, 2020-04-08:
@ArturGasparetto talvez ignorância minha, mas as normas relacionadas acima não estão detalhadas  no site da ABNT. Como ter a informação que precisamos?

buenow, 2020-04-08:
@ArturGasparetto RDC 349/2020: define critérios procedimentos extraordinários para regularização de equipamentos de proteção individual, de equipamentos médicos do tipo ventilador pulmonar e de outros dispositivos médicos identificados como estratégicos pela Anvisa.
Precisamos ver isso

guiriole, 2020-04-08:
Sim... tem que fazer todo o processo. Estou tentando vencer o passo um. 
Quem for fazer precisa utilizar o Internet Explorer v6 ou superior. Mas se for a versão 10 e 11 precisa habilitar o modo de compatibilidade.

Quanto a certificação de boas práticas, ha um relaxamento da exigência na RDC n 346, com a possibilidade de inspeção por videoconferência; e o art 8o permite um certificado temporário que ainda preciso estudar com mais calma

buenow, 2020-04-08:
@guiriole @ArturGasparetto A Anvisa vai priorizar a análise de pedidos de regularização de equipamentos de proteção individual (EPIs), ventiladores pulmonares e outros dispositivos médicos estratégicos para o enfrentamento do novo coronavírus (Covid-19). Para isso, serão adotados os critérios extraordinários e temporários da Resolução da Diretoria Colegiada (RDC) 349/2020, com validade a partir de sexta-feira (20/3), data de sua publicação no Diário Oficial da União (D.O.U.).   

Os produtos previstos na resolução são considerados estratégicos pela Anvisa para lidar com a situação de emergência de saúde pública internacional provocada pelo novo coronavírus. Por isso, o objetivo da medida é dar prioridade e mais rapidez aos processos de análise e regularização desses produtos.

buenow, 2020-04-08:
@guiriole @ArturGasparetto http://portal.anvisa.gov.br/noticias/-/asset_publisher/FXrpx9qY7FbU/content/simplificada-a-regularizacao-de-epis-e-outros-produtos/219201?p_p_auth=fY2YVI6E&inheritRedirect=false&redirect=http%3A%2F%2Fportal.anvisa.gov.br%2Fnoticias%3Fp_p_auth%3DfY2YVI6E%26p_p_id%3D101_INSTANCE_FXrpx9qY7FbU%26p_p_lifecycle%3D0%26p_p_state%3Dnormal%26p_p_mode%3Dview%26p_p_col_id%3D_118_INSTANCE_KzfwbqagUNdE__column-2%26p_p_col_count%3D2

ArturGasparetto, 2020-04-08:
@buenow Isso, a RDC 349 vai facilitar o trâmite de registro, porém, pelo que entendi, não exime o produto de passar por avaliação de conformidade.
Por não ser especialista na área de engenharia biomédica e menos ainda medicina, e também por nunca ter tido experiência na parte de avaliação de equipamentos eletromédicos e afins, ainda tenho dúvidas em relação a todas as normas e especificações que o ventilador precisa atender.
Eu não sei dizer ainda se aquela lista que o @OttoHeringer colocou lá em cima estaria completa com relação a cobrir todas normas e requisitos que o equipamento precisa atender.
Uma das formas de saber é estudar a legislação toda e o processo todo, como estamos fazendo, já que teremos que passar pelo processo.
Outra forma de se achar um atalho para obter a informação de tudo que precisa ser atendido, com relação à parte técnica do ventilador, talvez seja tentar entrar em contato com laboratórios acreditados no escopo de avaliação de conformidade desse tipo de equipamento eletromédico ou fabricantes que já passaram por esses processos, para ver se conseguimos que eles dêem essas informações mais rapidamente.

buenow, 2020-04-08:
@ArturGasparetto entendi. Vou te ajudar nisso. Assim que conseguirmos definir por onde seguir. Também não tenho experiência nessa área. Sou da área de desenvolvimento de hw e sw...

ArturGasparetto, 2020-04-08:
Deletei meu comentário pois agora que percebi que o artigo 7º da RDC 349 cita que "_Excepcionalmente, os produtos de que trata esta Resolução ficam dispensados de certificação no âmbito do Sistema Brasileiro de Avaliação da Conformidade (SBAC)._"
Desta forma, o ponto é entender quais são os requisitos mínimos necessários que se precisa cobrir para se colocar em uso o ventilador.
Eu também não sou especialista na área de engenharia biomédica nem medicina, sou da área de certificação em dispositivos de certificação digital, então neste contexto estou tendo que estudar o processo para saber quais normas são utilizadas como base para certificação e assim tentar determinar os requisitos técnicos.
Algumas normas já foram listadas pelo @OttoHeringer , mas precisamos saber se elas cobrem todo 
espectro que o equipamento precisa atender.
Talvez se conseguirmos contato com algum laboratório que realize esse tipo de avaliação de conformidade para ventiladores e equipamentos eletromédicos, possamos tentar obter alguma informação um pouco mais rápido. Contato com algum fabricante de respiradores/ventiladores pulmonares que tenha passado pelo processo de certificação também seria útil, mas acho que é um contato mais dífícil de se obter.
O LEB da Poli faz uma parte do escopo desses testes e vi que há colaboradores no projeto, mas não sei se eles estão acompanhando esta discussão.

ArturGasparetto, 2020-04-08:
Eu coloquei lá em cima o link para as normas gratuitas no site da ABNT que são descritas na IN 49/2019 do Ministério da Saúde ([IN 49 Ministério da Saúde-NOVEMBRO DE 2019.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4450886/IN.49.Ministerio.da.Saude-NOVEMBRO.DE.2019.pdf)) que estabelece a lista de Normas Técnicas, cujos parâmetros devem ser adotados para a certificação de conformidade, no âmbito do Sistema Brasileiro de Avaliação da Conformidade (SBAC), dos equipamentos sob regime de Vigilância Sanitária.

ArturGasparetto, 2020-04-08:
Acredito que se filtrarmos as normas listadas na IN 49 para o ventilador e outras normas derivadas ou referenciadas, teremos como listar também os requisitos técnicos.
Contudo, como disse antes, não sei se a lista feita anteriormente já é suficiente...tenho dúvida, por exemplo, se temos que considerar também normas que falem sobre ventiladores de transporte e emergência ou suporte ventilatório domiciliar, apesar do objetivo de contexto de uso do INSPIRE ser na UTI, pelo que entendi...
Neste contexto de auxiliar a entender se o espectro de normas a ser atendida é suficiente e quais requisitos mínimos podemos considerar para adequar o produto é que acredito que o apoio de um laboratório acreditado no escopo de ventiladores e equipamentos eletromédicos para avaliação de conformidade, ou de um fabricante deste tipo de produto, possa ser útil.

buenow, 2020-04-08:
@ArturGasparetto Excelente trabalho

buenow, 2020-04-08:
@ArturGasparetto vou ver em detalhes a IN 49 agora. Entendo que maior flexibilidade de uso melhor. As UTIs ficarão lotadas em pouco tempo. É comum improvisação de ambientes hospitalares em situações assim, bem como atendimento em trânsito, mesmo após a pandemia. Talvez devêssemos considerar isso.

ArturGasparetto, 2020-04-08:
Sim, concordo com você @buenow !
Contudo, precisamos entender o que será exigido como mínimo por comitês de ética e até mesmo órgãos como a própria ANVISA para liberar fabricação e uso destes aparelhos.
Outra preocupação, tentar saber a complexidade das exigências de forma a tentarmos realizar testes que possam atestar que estamos atendendo essas exigências...isso também é uma coisa difícil de se dizer no momento, se teremos condições de realizar os testes necessários para verificar o funcionamento sob certas condições ou aspectos.

ArturGasparetto, 2020-04-08:
Se não for possível saber de antemão esse mínimo exigido por órgão responsáveis, talvez no primeiro momento o melhor seja fazermos sugestões e depois ouvir comentários sobre essas sugestões de escopo de requisitos e testes a serem feitos...

buenow, 2020-04-08:
@ArturGasparetto, dê uma lida neste documento. É importante.
[Resolução da Diretoria Colegiada 349-2020.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4451319/Resolucao.da.Diretoria.Colegiada.349-2020.pdf)

guiriole, 2020-04-08:
Complementando essa RDC 349, vejam essa publicação da ANVISA abaixo no link. Iremos tentar por esse caminho.
http://portal.anvisa.gov.br/noticias/-/asset_publisher/FXrpx9qY7FbU/content/esclarecimentos-sobre-priorizacao-de-cbpf-e-produtos/219201

ArturGasparetto, 2020-04-08:
Sim, li este RDC também @buenow , e vi que você realçou algumas partes do texto.
Você tem alguma observação com relação a essas partes ou foi uma forma de indicar que seriam os pontos mais importantes deste RDC?

buenow, 2020-04-08:
apenas pontos que considero mais importantes pra esse caso, para meu entendimento apenas...

ArturGasparetto, 2020-04-08:
@guiriole Tenho uma dúvida quanto a classificação do equipamento. Qual critério o classifica como classe 3 e não classe 2? Pela RDC 185 eu estava achando que talvez fosse um equipamento classe 2...

guiriole, 2020-04-09:
> @guiriole Tenho uma dúvida quanto a classificação do equipamento. Qual critério o classifica como classe 3 e não classe 2? Pela RDC 185 eu estava achando que talvez fosse um equipamento classe 2...

Entendo que a regra 10 c), da RDC 185, da a entender ser classe III por se tratar de processos vitais com risco a vida. E nesse mesmo inciso a mera “monitoração” já é classe III. Acho que dá para concluir o que o “controle” da respiração minimamente seria classe III.

Também, em algumas pesquisas, identifiquei registros na Anvisa de ventiladores pulmonares na Classe III, mas não na Classe II

FilipeFer08, 2020-04-09:
Olá, boa noite, tudo bem?

Me chamo Filipe Ferreira, falo em nome da empresa Contourline Equipamentos Médicos e Estéticos, localizada em Sete Lagoas -MG.
Nosso diretor pediu-me que pudesse entrar em contato com vocês, com intenção de contribuir junto ao projeto Inspire OpenLung. Possuímos diversos equipamentos certificados por INMETRO e ANVISA, onde os mesmos são comercializados por todo o Brasil, possuímos um sistema de Produção, Assistência Técnica e Qualidade, onde aplicamos diversos normativas que podem interessar á vocês, incluindo algumas citadas acima. Peço que retorne esse contato, que possamos aprofundar um pouco mais no assunto e contribuir com o que pudermos. 

Enviei um Email para inspirepoliusp@gmail.com, agradeço o retorno.

guiriole, 2020-04-09:
@FilipeFer08 olá! Toda ajuda é sempre bem vinda. Vocês já comercializam essa linha de produto? Pretendem? Acho que, nesse tópico de discussão, estamos entendendo que o passo eh fazer diretamente o registro do produto (conforme comentários anteriores), na Classe III. Poderia nos ajudar na compreensão dessa etapa?

ArturGasparetto, 2020-04-09:
> Entendo que a regra 10 c), da RDC 185, da a entender ser classe III por se tratar de processos vitais com risco a vida. E nesse mesmo inciso a mera “monitoração” já é classe III. Acho que dá para [...]

Entendi @guiriole !
Faz sentido mesmo.
Obrigado pela explicação!

ArturGasparetto, 2020-04-09:
Pessoal, vou encaminhar aqui uma informação obtida junto a Gerência de Tecnologia em Equipamentos (GQUIP) da ANVISA que pode auxiliar e também responder possíveis perguntas:

"**Orientações regulatórias relacionadas à submissão de registro junto à Anvisa de ventiladores pulmonares**

Encaminhamos, anexo, material para que os projetistas de ventiladores pulmonares sejam auxiliados.
A seguir, breve apresentação desses materiais.

Anexo 1. Documento elaborado pelo MHRA (UK), entidade regulatória da Inglaterra, que contém requisitos/especificações/parâmetros de desempenho clínico minimamente aceitáveis para os ventiladores pulmonares básicos. Trata-se de um documento bastante completo. 
[Anexo-1-MHRA-Basic-Ventilator.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4458297/Anexo-1-MHRA-Basic-Ventilator.pdf)

Anexos 2, 3, 4. Documentos disponibilizados pelo ECRI, um instituto não governamental que traz diversos estudos e informações sobre segurança e qualidade mínima de equipamentos. O material inclui artigo sobre ventiladores para cuidado intensivo, foco da demanda atual para o Covid-19 (Anexo 2) e artigo sobre segurança de ventiladores (Anexo 3). A fim de auxiliar, o arquivo anexo denominado “Ventiladores para ambiente de cuidado intensivo – recomendações ECRI” foi preparado para compilar as informações do ECRI (Anexo 4) e exemplificar parâmetros implementados em um ventilador pulmonar nacional, disponibilizado no mercado.
[Anexo-2-ECRI-Ventilators-Intensive-Care.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4458311/Anexo-2-ECRI-Ventilators-Intensive-Care.pdf)
[Anexo-3-ECRI-Ventilators-Safety.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4458320/Anexo-3-ECRI-Ventilators-Safety.pdf)
[Anexo-4-Ventiladores-ambiente-cuidado-intensivo-recomendações-ECRI-compiladas.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4458321/Anexo-4-Ventiladores-ambiente-cuidado-intensivo-recomendacoes-ECRI-compiladas.pdf)

Ainda para auxílio quanto ao projeto, deve-se verificar a ABNT NBR ISO 80601-2-12:2014 (disponibilizada gratuitamente pelo link (https://www.abntcatalogo.com.br/norma.aspx?ID=326196#). Apesar de não haver necessidade de certificação Inmetro para aceitar a regularização, os projetos devem ser conduzidos sob tais requisitos de segurança básica e desempenho essencial.

Com relação aos requisitos essenciais para eletromédicos, observar as normas gerais cabíveis da série ABNT NBR 60601 (também disponibilizada gratuitamente pela ABNT). 

Citamos a RDC 16:2013, embasada na ISO 13485, que traz os requisitos relacionados às boas práticas de fabricação de produtos para a saúde (http://bvsms.saude.gov.br/bvs/saudelegis/anvisa/2013/rdc0016_28_03_2013.pdf). Observar que esse regulamento trata da necessidade de se manter, de forma orgânica e contínua, o Gerenciamento de Risco, que envolva todo o ciclo de vida do produto, desde sua concepção até a descontinuação. Ademais, deve-se manter o registro dos fornecedores de componentes e materiais que comporão o produto, qualificando-os de acordo com a criticidade desses componentes e materiais, bem como provendo sua devida rastreabilidade. É necessário observar todos os requisitos aplicáveis que constam nesse regulamento.

Por fim, resumimos basicamente o que avaliamos em um registro de equipamentos:

- Classificação de risco: Os ventiladores pulmonares são dispositivos de alto risco, classificados na Classe III da RDC 185/2001.
- Modelos de rotulagem e instruções de uso são apresentados.
- Deverá ser apresentado um Relatório Técnico, em que constem os itens descritos na RDC acima citada. Com relação ao item 1.6 do RT, sobre necessidade de descrever os requisitos essenciais de eficácia e segurança, tais requisitos também deverão ser comprovados – seja à partir de ensaios embasados em metodologias consagradas, à partir de relatório comparativo relacionado a dispositivos semelhantes que já estejam no mercado (neste caso comprovar a semelhança durante o desenvolvimento do projeto). 
- Caso os requisitos essenciais de eficácia e segurança não sejam comprovados, deverá ser apresentada pesquisa clínica do uso do dispositivo em pacientes. 
- Ainda no item de requisitos essenciais de eficácia e segurança, é importante apresentar uma FMEA (análise de modo e efeito de falha).
- A parte referente aos requisitos essenciais é de especial importância porque é ali que o fabricante garante a eficácia do produto e sua segurança – seja elétrica, seja com relação à usabilidade, à robustez do sistema e à eficácia e segurança clínica. Fazemos um adendo específico para os projetos de ventiladores pulmonares que foram compostos por componentes críticos que não são comumente utilizados nos dispositivos de mercado, por ex., o uso de “ambus”. É de primordial importância que se avalie clinicamente, sob auxílio de especialistas habituados à prática intensivista e à pneumologia, a razão de risco/benefício de uso desses dispositivos, bem como o período admissível para se manter o paciente conectado. 
- Detalhe importante a respeito da submissão à Anvisa desses registros: a pessoa jurídica que submete o processo deve estar habilitada quanto à certificação de Boas Práticas de Fabricação de produtos de classe de risco III. Deste modo, mesmo que uma empresa que atue em outro ramo de produção projete um dispositivo destes, é necessário que o processo de regularização pela Anvisa seja submetido por uma empresa que possua CBPF. Vejam que o processo  [... cut ...]
- Caso uma empresa, que não possua CBPF, tiver interesse em registrar, por si mesma, um dispositivo de uso médico de classe de risco III ou IV, poderá solicitar sua certificação junto à CPROD/GGFIS/Anvisa. A área está priorizando a certificação das empresas que possuam interesse em fabricar os dispositivos críticos em caráter emergencial.

Reforçamos que estamos empenhados em prover auxílio regulatório e, não menos importante, que sejam asseguradas a segurança e eficácia desses novos dispositivos em desenvolvimento."

ArturGasparetto, 2020-04-09:
@OttoHeringer e @emersonmoretto Baseado nas informações que o pessoal da GQUIP da ANVISA forneceu, o pessoal do LSI-TEC LEA e eu iremos estudar de forma mais focada as normas ABNT NBR ISO 80601-2-12:2014 e ABNT NBR IEC 60601, para também já começarmos a elaborar um documento de sugestões de requisitos técnicos mínimos a serem atendidos, baseados nas normas citadas.
Pensamos em ir disponibilizando esse documento ao longo da sua elaboração, permitindo discussões, sugestões e também avaliações.
O objetivo é permitir que haja uma avaliação desse documento de requisitos também por especialistas e, principalmente, da ANVISA. Se estes especialistas e ANVISA avalizarem nossa sugestão de conjunto mínimo de requisitos técnicos, saberemos como teremos que atendê-lo e poderemos verificar como testar e, por fim, atestar o funcionamento do ventilador sob esse conjunto de requisitos, gerando um Relatório Técnico que será utilizado no processo de registro e autorização de uso do equipamento.

buenow, 2020-04-09:
@ArturGasparetto Gostaria de ajudar nesse processo de adequação.  O que você conseguiu é muito valioso. Parabéns pela iniciativa!

buenow, 2020-04-09:
Parabéns pela iniciativa. É realmente um material importante este que você conseguiu, bem como o contato.

ArturGasparetto, 2020-04-09:
> 
> 
> Parabéns pela iniciativa. É realmente um material importante este que você conseguiu, bem como o contato.

Na verdade os parabéns tem que ser dado ao @emersonmoretto que conseguiu essa informação, o contato e permitiu que eu publicasse aqui, pois tinha visto que poderia ajudar bastante...como já ajudou a definirmos um escopo mínimo de normas para iniciar a levantar os requisitos técnicos.

buenow, 2020-04-09:
Excelente. Posso ajudar?

OttoHeringer, 2020-04-10:
> @OttoHeringer e @emersonmoretto Baseado nas informações que o pessoal da GQUIP da ANVISA forneceu, o pessoal do LSI-TEC LEA e eu iremos estudar de forma mais focada as normas ABNT NBR ISO [...]

É exatamente isso que penso também. Se listarmos tudo podemos falar com os órgãos de regulamentação de maneira mais propositiva, sendo específicos. Caso identifiquemos que algumas normas não sejam exatamente necessárias, sem que compromentam de maneira crítica o bom funcionamento e segurança do respirador, podemos sugerir a simplificação dos critérios para deixar mais viável regularizar isso a tempo.

@FilipeFer08 , obrigado por se disponibilizarem. Aqui é o canal mais rápido de interagir. Caso vocês tenham interesse em ajudar na parte de escalonamento da produção, encaminharei seu email para o responsável; agora caso queira contrinbuir com essa questão regulatória, sugiro que contribua por este "fórum" aqui mesmo, pois pode ajudar mais pessoas. Como o projeto é aberto, já existem iniciativas replicando o projeto e que poderiam entrar em contato com vocês por aqui, por exemplo. A equipe de especialistas nessa parte regulatória de vcs poderia validar o que estamos discutindo até agora?

OttoHeringer, 2020-04-10:
> Adicionei os links do site do catálogo da ABNT para as normas que foram listadas pelo @OttoHeringer . Todas essas normas encontram-se como gratuitas no momento.
> 
> 1. ABNT NBR IEC 60601-1:2010 - Equipamento eletromédico - Parte 1: Requisitos gerais para segurança básica e desempenho essencial [https://www.abntcatalogo.com.br/norma.aspx?ID=80437](url)
> 2. ABNT NBR IEC 60601-1:2010  Emenda 1:2016 - Equipamento eletromédico - Parte 1: Requisitos gerais para segurança básica e desempenho essencial [https://www.abntcatalogo.com.br/norma.aspx?ID=355495](url)
> 3. ABNT NBR IEC 60601-1-2:2017 - Equipamento eletromédico - Parte 1-2: Requisitos gerais para segurança básica e desempenho essencial - Norma Colateral: Perturbações eletromagnéticas - Requisitos e ensaios [https://www.abntcatalogo.com.br/norma.aspx?ID=383029](url)
> 4. ABNT NBR IEC 60601-1-6:2011 Versão Corrigida 2013 - Equipamento eletromédico - Parte 1-6: Requisitos gerais para segurança básica e desempenho essencial - Norma colateral: Usabilidade [https://www.abntcatalogo.com.br/norma.aspx?ID=307130](url)
> 5. ABNT NBR IEC 60601-1-8:2014 Versão Corrigida 2015 - Equipamento eletromédico - Parte 1-8: Requisitos gerais para segurança básica e desempenho essencial - Norma colateral: Requisitos gerais, ensaios e diretrizes para sistemas de alarme em equipamentos eletromédicos e sistemas eletromédicos [https://www.abntcatalogo.com.br/norma.aspx?ID=348135](url)
> 6. ABNT NBR IEC 60601-1-10:2010  Versão Corrigida 2013 - Equipamento eletromédico - Parte 1-10: Requisitos gerais para segurança básica e desempenho essencial - Norma colateral: Requisitos para o desenvolvimento de controladores fisiológicos em malha fechada [https://www.abntcatalogo.com.br/norma.aspx?ID=307136](url)
> 7. ABNT NBR IEC 60601-1-10:2010 Emenda 1:2017 - Equipamento eletromédico - Parte 1-10: Requisitos gerais para segurança básica e desempenho essencial — Norma colateral: Requisitos para o desenvolvimento de controladores fisiológicos em malha fechada [https://www.abntcatalogo.com.br/norma.aspx?ID=371257](url)
> 8. ABNT NBR ISO 80601-2-12:2014 - Equipamento eletromédico - Parte 2-12: Requisitos particulares para a segurança básica e o desempenho essencial de ventiladores para cuidados críticos [https://www.abntcatalogo.com.br/norma.aspx?ID=326196](url)
> 
> Obs.: _As normas técnicas listadas acima foram aprovadas pela Instrução Normativa do Ministério da Saúde nº 49 de 27 de Novembro de 2019 ([IN 49 Ministério da Saúde-NOVEMBRO DE 2019.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4447684/IN.49.Ministerio.da.Saude-NOVEMBRO.DE.2019.pdf)), cujos parâmetros devem ser adotados para a certificação de conformidade, no âmbito do Sistema Brasileiro de Avaliação da Conformidade (SBAC), dos equipamentos sob regime de Vigilância Sanitária._
> 
> 1. ABNT NBR ISO 14971:2009 - Produtos para a saúde — Aplicação de gerenciamento de risco a produtos para a saúde [https://www.abntcatalogo.com.br/norma.aspx?ID=55540](url)
> 2. ABNT NBR ISO TR 24971:2014 - Produtos para a saúde - Orientações para a aplicação da ABNT NBR ISO 14971 [https://www.abntcatalogo.com.br/norma.aspx?ID=325424](url)
> 3. ABNT NBR IEC 62366: 2016 - Produtos para a saúde — Aplicação da engenharia de usabilidade a produtos para a saúde [https://www.abntcatalogo.com.br/norma.aspx?ID=354941](url)

Boa @ArturGasparetto ! Vou transformar isso num documento da wiki, junto com os comentários do @guiriole  e o @buenow ; é a melhor síntese da discussão até aqui. Obrigado, vcs são demais!

buenow, 2020-04-10:
@OttoHeringer Olá Heringer, meu pessoal gostaria de ajudar no escalonamento da produção. Creio que precisaremos ordenar as informações do processo produtivo para tanto. Você pode nos ajudar? Obrigado.

OttoHeringer, 2020-04-11:
> @OttoHeringer Olá Heringer, meu pessoal gostaria de ajudar no escalonamento da produção. Creio que precisaremos ordenar as informações do processo produtivo para tanto. Você pode nos ajudar? Obrigado.

Me mande uma mensagem no Telegram!

OttoHeringer, 2020-04-11:
Pessoal, e a norma ABNT NBR ISO 10651-3:2014  "Ventiladores pulmonares para uso médico – Parte 3: Requisitos particulares para ventiladores de transporte e emergência"? Não se aplicaria nem um pouco no nosso caso!?

DeyviStecker, 2020-04-13:
@OttoHeringer , O Instituto Federal de Ciência e Tecnologia do Mato possui interesse em cooperar na Produção , levando o projeto da UsP para o Centro Oeste do Brasil. Uma equipe da engenharia de controle e automação está disposta.

OttoHeringer, 2020-04-14:
oi @DeyviStecker, obrigado por querer ajudar nessa parte.
Como isso já foge do "meu departamento", o caminho ideal é você mandar um email para o responsável por essa parte, o Dário Gramorelli: dario@alfametal.com.br
Como você é ligado a uma instituição de ensino e pesquisa, pode me mandar uma mensagem pelo Telegram? Você consegue me encontrar pelo chat do grupo de discussão desse repositório: https://t.me/openlungpoliusp

AlvaroAguiar, 2020-04-14:
> Entendi. Pensei que fast track pudesse significar mais coisas além disso que você mencionou.
>
> Sim, já existe uma conversa com alguns profs e a ANVISA. Apesar de possivelmente o tempo de apreciação [...]

Otto, seria extremamente importante que esse projeto pudesse ser replicado em QUALQUER LOCAL, por QUALQUER PESSOA, tendo a importância que tem! Meu interesse mesmo é o de produzi-lo na minha cidade e oferecê-lo às unidades de atendimento hospitalar, Talvez cobrando preço de custo de produção, sem interesse financeiro.

[12 more replies]