Inspire-Poli-USP/Inspire-OpenLung #15: Especificação do motor

karlaycosta, 2020-04-01:
Olá pessoal gostaria de saber sobre a especificação do motor de passo, observei que nos testes https://github.com/Inspire-Poli-USP/Inspire-OpenLung/blob/master/Teste_Poli_USP_LEB_26_03_2020 a especificação está descrito 20kg e na atualização do dia "24 de Março: Modelagem 3D e testes do Protótipo V3: agora projetado para motor NEMA 23 de 15kg de torque" esta descrito com 15kg, seria um NEMA 23 de 15kg ou 20kg de torque?

emersonmoretto, 2020-04-01:
Opa, o de 15 é suficiente

Usamos o de 20 também porque é o que tínhamos em mais quantidade

Vou atualizar no git, valeuu
*Emerson Moretto*
emerson@moretto.eng.br
11 99484-0571

alberiolima, 2020-04-02:
Segundo a imagem, num ciclo de 3 segundos, a inspiração, etapa 1 do ciclo, deve levar 1 segundo.
Mas calculando com um curso de 100mm (10cm), usando um motor de passo, com 200 passos por revolução, em usar micropassos é claro, com uma barra roscada com passo 2mm. Teríamos 100passos por mm. Para percorrer 100mm são necessários 10.000passos. Então a velocidade seria de 10.000passos/segundo. E a biblioteca que controla o motor tem a informação que velocidada acima de 1.000 não é confiável, imagina 10.000

[image]

HPparanhos, 2020-04-02:
@alberiolima É quase impossível replicar esse projeto com fuso de passo 2mm. Estamos utilizando aqui um de 25mm e @emersonmoretto um de 15mm. Com isso o avanço do sistema se torna ideal com poucos passos e o motor. Esqueça neste caso de usar micropassos! Isso reduz ainda mais o torque do motor.

alberiolima, 2020-04-02:
> @alberiolima É quase impossível replicar esse projeto com fuso de passo 2mm. Estamos utilizando aqui um de 25mm e @emersonmoretto um de 15mm. Com isso o avanço do sistema se torna ideal com poucos passos e o motor. Esqueça neste caso de usar micropassos! Isso reduz ainda mais o torque do motor.

Ótimo então! Quanto a aceleração acredito que não tem necessidade, para controlar o fluxo/vazão só seria necessário ter uma tabela ou equação que retorne o valor de ml para cada posição no curso!

alberiolima, 2020-04-02:
@HPparanhos, o @emersonmoretto falou que está testando uma versão que usa PID, mas não entendi a necessidade! Você também está testando?!

HPparanhos, 2020-04-02:
PID servirá para controle da curva de respiração. Mas existem algumas dificuldades em atingir essa curva usando sistema pid, pois após "overshoot" ele continuará fazendo auto compensação com leitura dos sensores, o que pode trazer valores diferentes do esperado. Estamos em busca também de outro tipo de controle para que essa curva seja respeitada. 

Nosso foco está na validação de IHM e controle das válvulas no momento, nosso FORK já está atualizado com ultimas modificações.

andreribeiroeco, 2020-04-03:
Olá pessoal, no projeto (estou analisando a V5) tem uma peça que parece uma arruela de acrílico, que serve de guia para a peça móvel. Na pasta de cortes 2D não tem o DXF dela, aí vem a pergunta. Ela realmente é uma peça cortada em acrílico e esqueceram de gerar o DXF dela ou ela é outra peça e como está tudo a mesma cor eu me confundi? Obrigado.

karlaycosta, 2020-04-03:
Fiquei com essa dúvida também, parece que a peca foi impressa em 3D? Estou pensando em adaptar usando micro rolamento com flange tipo **F624zz**, levando em consideração a resistência e durabilidade do rolamento, creio que o @emersonmoretto possa tirar essa dúvida quanto ao material usado na peça que parece uma bucha para ajudar a deslizar a parte móvel deslocada pelo motor, o problema em usar o rolamento citado, seria na dificuldade em encontrar nas lojas!

alberiolima, 2020-04-04:
> PID servirá para controle da curva de respiração. Mas existem algumas dificuldades em atingir essa curva usando sistema pid, pois após "overshoot" ele continuará fazendo auto compensação com leitura [...]

Quais válvulas?

emersonmoretto, 2020-04-04:
> Olá pessoal, no projeto (estou analisando a V5) tem uma peça que parece uma arruela de acrílico, que serve de guia para a peça móvel. Na pasta de cortes 2D não tem o DXF dela, aí vem a pergunta. Ela [...]

é uma peça cortada a laser de PTFE, de 25mm de diametro e furo de 3mm. Mas ainda estamos avaliando se ela realmente resolve a questão do movimento linear. Está há 2 dias rodando sem parar aqui e até agora parece bom, porém há muito desgaste já. Provavelmente tenhamos que mudar pra algum material mais resistente.

alberiolima, 2020-04-11:
Fiz um código para teste de motor de passo, usando a biblioteca accelStepper, apesar de não recomendar, ela é visivelmete simples! 
https://youtu.be/Z4LCoNhKEyM 
[image]

Gustavomurta, 2020-04-18:
Srs, já divulguei no grupo do Telegram.
Estamos fusando a Biblioteca **FlexyStepper**. No meu ponto de vista, tem recursos mais avançados do que a AccelStepper. 
O controle dos movimentos são fantásticos e precisos! Defino a distancia percorrida em mm, velocidade (mm/seg) , aceleração (mm/seg2).  E enquanto o motor gira, posso testar algumas variáveis. 
https://github.com/Stan-Reifel/FlexyStepper

https://github.com/tiagocriaar/AMBU-SOL-AR/blob/master/C%C3%B3digos%20Arduino/SW_RESPIRADOR_SOL_AR_V1_40a.ino
Obs - o código esta bagunçado pois outro colega esta implementando outras funções. E estamos fazendo tudo às pressas. 

Gustavo Murta 
Anbu Sol e Ar

Gustavomurta, 2020-04-18:
Meus cálculos de RPM, usando fuso de 25 mm. 

[image]

alberiolima, 2020-04-18:
> Srs, já divulguei no grupo do Telegram.
> Estamos usando a Biblioteca **FlexyStepper**. No meu ponto de vista, tem recursos mais avançados do que a antiga AccelStepper.
> O controle dos movimentos são fantásticos e precisos! Defino a distancia percorrida em mm, velocidade (mm/seg) , aceleração (mm/seg2). E enquanto o motor gira, posso testar algumas variáveis.
> https://github.com/Stan-Reifel/FlexyStepper
> 
> https://github.com/tiagocriaar/AMBU-SOL-AR/blob/master/C%C3%B3digos%20Arduino/SW_RESPIRADOR_SOL_AR_V1_40a.ino
> Obs - o código esta bagunçado pois outro colega esta implementando outras funções. E estamos fazendo tudo às pressas.
> 
> Gustavo Murta
> Ambu Sol e Ar

> Srs, já divulguei no grupo do Telegram.
> Estamos usando a Biblioteca **FlexyStepper**. No meu ponto de vista, tem recursos mais avançados do que a antiga AccelStepper.
> O controle dos movimentos são fantásticos e precisos! Defino a distancia percorrida em mm, velocidade (mm/seg) , aceleração (mm/seg2). E enquanto o motor gira, posso testar algumas variáveis.
> https://github.com/Stan-Reifel/FlexyStepper
> 
> https://github.com/tiagocriaar/AMBU-SOL-AR/blob/master/C%C3%B3digos%20Arduino/SW_RESPIRADOR_SOL_AR_V1_40a.ino
> Obs - o código esta bagunçado pois outro colega esta implementando outras funções. E estamos fazendo tudo às pressas.
> 
> Gustavo Murta
> Ambu Sol e Ar

Mas como a maneira de controle de passos é a mesma da AccelStepper, continua o problema de não poder executar nada enquando o motor se move, e sabemos que é necessário ler sensores e efetuar cálculos. Isso pode ser resolveido com RTOS, PWM, deixar o controle do motor num processador diferente!

Gustavomurta, 2020-04-18:
Opa!!! "continua o problema de não poder executar nada enquando o motor se move"
Isso não precede! Veja o meu comentário:

E enquanto o motor gira, posso testar algumas variáveis.

alberiolima, 2020-04-18:
> Opa!!! "continua o problema de não poder executar nada enquando o motor se move"
> Isso não precede! Veja o meu comentário:
> 
> E enquanto o motor gira, posso testar algumas variáveis.
> 
> https://github.com/Stan-Reifel/FlexyStepper/tree/master/examples/Example7_MovingWhileDoingOtherThingsToo

Coloque calculo com variaveis float ou double

Gustavomurta, 2020-04-18:
A minha ideia é usar o tempo de delay do Plato, para fazer isso... 
Não testei ainda. O que acha ?

Gustavomurta, 2020-04-18:
Concordo que é um desafio, mas estou confiante que conseguirei.

alberiolima, 2020-04-18:
> Concordo que é um desafio, mas estou confiante que conseguirei.

Uma *gambiarra seria executar "stepper.processMovement();" numa interrupção por tempo

alberiolima, 2020-04-18:
Uma solução +/- seria separar o controle do motor do restante do sistema, já que isso é crítico!
poderia fazer comunicação entre os módulos usando SPI que é uma comunicação muito rápida, já tem implementada no hardware da muitos microcontroladores!

Gustavomurta, 2020-04-19:
Alberio, muito obrigado pelas dicas! 
Se o Arduino não aguentar, não hesitarei em sugerir mudança para o ESP32.
Dois processadores rodando 10 vezes o clock de um Arduino. (até 240 MHz)

alberiolima, 2020-04-19:
> Alberio, muito obrigado pelas dicas!
> Se o Arduino não aguentar, não hesitarei em sugerir mudança para o ESP32.
> Dois processadores rodando 10 vezes o clock de um Arduino. (até 240 MHz)

Acho que um STM32F103 já resolveria, com 20Kb de RAM, e 64Kb ou 128Kb de flash com clock de 72mhz, acredito que já seria suficiente, pode continuar usando a IDE do Arduino. Usando FreeRTOS. (bluepill)
Evitaria o ESP32 pelo ADC ser um pouco ruim de usar, e pelo fato de a maioria dos módulos prontos ter antena de RF, isso talez seja um problema para validar! Com certeza você teria um grande porder de processamento, muita RAM e muita FLASH, mais que o necessário!

Tyrtech, 2020-05-03:
Já fizeram testes de longa duração? eu uso esses motores a muito tempo em maquinas cnc, e eles queimam se forem usados por um longo período, os meus já queimaram com 4 horas de uso, ai coloquei um cooler atrás que resolveu o problema, nunca testei com dissipador de calor, vocês estão pensando em um método de resfriamento pro motor ou não tiveram problemas?

Gustavomurta, 2020-05-04:
Tyrtech, 
Se o motor da sua CNC, esta queimando depois de 4 horas de uso, é porque tem algum problema na sua CNC. Ou você esta sobrecarregando os motores além do que eles suportam. Ou pode até ser que seja baixa qualidade do motor (chinês). Use marcas de boa qualidade e sugiro que reclame com o  fabricante da sua CNC. 
Eu trabalhei por muitos e muitos anos com manutenção de impressoras de grande porte, que usavam motores de passo gigantes (que puxavam os formulários). Essas impressoras IBM eram usadas o dia inteiro e dentro desse período de muitos anos, troquei uma vez o motor!!
Os motores de passo são super resistentes, e além do mais, os drivers atuais tem proteção contra sobre-corrente!

Gustavomurta, 2020-05-04:
Se alguém se interessar, criei esse Tutorial no Lab de Garagem, da montagem de um Sensor Hall para medições do RPM do motor do eixo sem fim do nosso Ambu Sol e Ar. Usando rotor de motor sem escovas BLDC, poderá fazer medições precisas dos movimentos do Respirador. Registrando todos os pulsos com um analisador lógico, poderá fazer aferições precisas de tempo de inspiração, tempo de platô e tempo de expiração. 
http://labdegaragem.com/forum/topics/sensor-hall-de-rota-o-rpm

Nessa imagem, consta os pulsos do motor durante o Tempo de inspiração, e os pulsos do motor para retorno do motor. 
 
[image]