Inspire-Poli-USP/Inspire-OpenLung #23: Especificação do sensor de pressão

m-rodrigues-souza, 2020-04-03:
Qual sensor de pressão está sendo utilizado nos testes? Tem o part number?

OttoHeringer, 2020-04-03:
Oi Matheus.
Boa pergunta. Nesse momento, para fazer os testes do algoritmo de controle, estamos medindo a pressão com equipamentos "padrão" que se usam associados a pulmões mecânicos, como esse, por exemplo:
[image]

Se você conseguir uma maneira confiável de fazer sensoriamento de pressão, acho que já "tá valendo" pra fazer os testes de controle.

Agora, o componente em si de controle, o @emersonmoretto estava pensando em usar um muito comum usado em carros.

Vou ver aqui e atualizo esse issue.

Mas se você quiser, tbm pode sugerir um sensor. Lembre-se que ele tem que ser barato e acessível "a pronta entrega", e em "qualquer lugar".

karlaycosta, 2020-04-03:
No momento estou utilizando o GYBMEP (pressão, temperatura e umidade) para testes aqui, um sensor muito confiável, porém, ainda estou analisando como acoplar ao sistema de mangueiras junto ao ambu.
Link com a descrição: [https://produto.mercadolivre.com.br/MLB-1200500296-gy-bme280-saia-temperatura-sensor-umidade-de-alto-exatido-_JM?quantity=1](url)

alberiolima, 2020-04-03:
> No momento estou utilizando o GYBMEP (pressão, temperatura e umidade) para testes aqui, um sensor muito confiável, porém, ainda estou analisando como acoplar ao sistema de mangueiras junto ao ambu. [...]

Conheço bem os sensores BMP085, BMP180,BMP280 e BME280. Uso para calculo de velocidade vertical e altitude, são bem confiáveis.
Mas seu range deve ser levando em consideração: 300hPa a 1100hPa. Pois a pressão atmosférica ao nível do mar é 1013,25hPa.
O range útil, para o fim, seria 1013,25hPa a 1100hPa, resultando 86,75hPa que corresponde a 88,46cmH2O. Mas no III Consenso Brasileiro de Ventilação Mecânica, Tabela 1, diz que a pressão máxima inspiratória é de 80 a 120cmH2O. Isso não seria um problema?

alberiolima, 2020-04-03:
Para usar sensor que não seja diferencial é necessário ler a pressão antes de iniciar a operação, para usar como referência, só que a pressão atmosférica é dinâmica ela muda durante o dia por vários motivos. Será que isso também não seria ruim? Acredito que seja necessário usar um sensor diferencial ou dois sensores.

karlaycosta, 2020-04-03:
Vou olhar o datasheet, se não estou enganado podemos alterar o valor de referência da pressão atmosférica (1013,25hPa) para fazer a calibração do sensor, assim poderíamos atingir o range (80 a 120cmH2O) com folga. Lembando que o sensor ficaria confinado dentro de uma válvula com o ambu, assim não levaria em consideração a pressão atmosférica e sim a pressão da tubulação, mas ainda estou analisando essa adaptação e como faria o confinamento do sensor, @alberiolima obrigado pelo feedback, estou olhando um sensor diferencial para comprar pois até a adaptação do sensor é mais fácil.
Link com a descrição: [https://www.usinainfo.com.br/sensor-de-pressao-arduino/sensor-de-pressao-diferencial-mpx5700dp-0kpa-a-700kpa-3812.html](url)

alberiolima, 2020-04-03:
> Vou olhar o datasheet, se não estou enganado podemos alterar o valor de referência da pressão atmosférica (1013,25hPa) para fazer a calibração do sensor, assim poderíamos atingir o range (80 a [...]

Separei dois modelos para analisar melhor:  (acredito que precisaremos de pressão negativa,também)
Sensor proximo do ideal XGZP6847040KPGPN -40KPa a 40KPa (-408cmH2O a 408cmH2O)
  XGZP6847020KPG (0 a 20Kpa)

buenow, 2020-04-05:
Pessoal, caso ainda não tenho decidido o sensor de pressão, seria o caso de considerarmos essa proposta?
[lps27hhw_sensor_pressao.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4434926/lps27hhw_sensor_pressao.pdf)

buenow, 2020-04-05:
Esta é uma proposta de uso possível...
[Presure measure.docx](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4434930/Presure.measure.docx)

alberiolima, 2020-04-05:
> Esta é uma proposta de uso possível...
> [Presure measure.docx](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4434930/Presure.measure.docx)

Muito bom, eu estava procurando um projeto assim que vi no passado, quando pesquisar sobre sensores de velocidade de vento, até tendei replicar, mas acabei usando esse:
https://www.banggood.com/3pcs-Digital-Barometric-40KPa-Pressure-Sensor-Module-Liquid-Water-Level-Controller-Board-p-1590571.html?cur_warehouse=CN

alberiolima, 2020-04-05:
Vou tentar montar uma câmara de vacuo para calibrar os sensores! Tenho alguns modelos, que uso em altímetros barométricos.
[image]

VitorFrost, 2020-04-06:
Como a pressão com certeza não será igual em todo o sistema, talvez seja necessário nessa primeira etapa colocar o sensor tanto na entrada (ambu) e outro no final (paciente). Fazer um levantamento de perda de pressão. O que vai alterar a pressão no sistema: altura entre o paciente e o ambu, VÁLVULAS, FILTROS, conectores, curvas das mangueiras, etc. Ou seja, tudo. Em um equipamento "tradicional" aonde é feita a leitura de pressão?
A pergunta mais importante! Na etapa de validação pela Anvisa, como os equipamentos são validados? O que é acoplado? Eles testam com os filtros HME ou HEPA? Assim como esse projeto sofrerá perdas, os equipamentos consagrados também sofrem e eu suspeito que ele mostra uma leitura nas condições ideais do teste, mas que no uso há uma pequena imprecisão.

OttoHeringer, 2020-04-14:
https://github.com/Inspire-Poli-USP/Inspire-OpenLung/issues/43#issuecomment-612534274

@danilochui encontoru uma referência de outro sensor da BOSCH que poderia ser usado

Gustavomurta, 2020-04-18:
Alguém já fez testes com esse sensor ? Comprei um hoje pelo ML. 
Mp3v5010dp 
https://www.nxp.com/docs/en/data-sheet/MP3V5010.pdf
0 to 10 kPa (0 to 1.45 psi)
0.1 to 3.1 V Output 

Gustavo Murta 
Ambu Sol e Ar

alberiolima, 2020-04-18:
> Alguém já fez testes com esse sensor ? Comprei um hoje pelo ML.
> Mp3v5010dp
> https://www.nxp.com/docs/en/data-sheet/MP3V5010.pdf
> 0 to 10 kPa (0 to 1.45 psi) ou (0 a 101,972 cm H2O)
> 0.1 to 3.1 V Output
> 
> Application Examples
> • Hospital Beds
> • HVAC
> • Respiratory Systems
> • Process Control
> 
> Gustavo Murta
> Ambu Sol e Ar

Nunca usei esse aí, mas usei alguns parecidos, resolução de 12bits, e comunicação SPI, boa resolução com ótima velocidade de comunicação! O próprio sensor já tendo um ADC já melhora muita coisa! Acredito que será uma boa opção!

alberiolima, 2020-04-18:
> Alguém já fez testes com esse sensor ? Comprei um hoje pelo ML.
> Mp3v5010dp
> https://www.nxp.com/docs/en/data-sheet/MP3V5010.pdf
> 0 to 10 kPa (0 to 1.45 psi) ou (0 a 101,972 cm H2O)
> 0.1 to 3.1 V [...]

Coloca o link do mercadolivre

Gustavomurta, 2020-04-19:
Alberio, acho que existe alguma restrição sobre colocar links comerciais aqui. 
Pesquise por Mp3v5010dp no ML. 
Corrigindo : 
A saída desse sensor é um sinal analógico - 0 a 3,1V .
Ele não tem um ADC interno. Seria ótimo se tivesse.

alberiolima, 2020-04-19:
> Alberio, acho que existe alguma restrição sobre colocar links comerciais aqui.
> Pesquise por Mp3v5010dp no ML.
> A saída desse sensor é um sinal analógico - 0 a 3,1V .
> Ele não tem um ADC interno. Seria ótimo se tivesse.
> Vi que o módulo que indicou tem um ADC de 12 bits. Muito bom.

Que pena pelo ADC, estava pensando no módulo pronto.

VitorFrost, 2020-04-23:
Esse projeto pode ser útil: https://hackaday.io/project/170622-visp-ventilator-inline-sensor-package

alexaquino, 2020-04-23:
Pessoal, a uma empresa nacional que pode produzir o sensor dentro do range necessário (faixa de 0 - 60CmH2O e saída indo de 4 a 20mA). Ele já nos forneceu um para um projeto e poderia nos fornecer algumas amostras para testes no Inspire. Caso haja interesse posso colocar ele em contato com o @OttoHeringer pra ele realizar o envio.

[Manual NP400.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4522630/manual_np400_v10x_d_portugues.pdf)

[image]

VitorFrost, 2020-04-23:
@alexaquino Muito bom conseguir um fornecedor nacional!
Verifiquei que no datasheet que o sensor funciona com 12V e mede-se a corrente de saída. O fabricante deu algum circuito básico de leitura? Uma simples ponte de Wheatstone é suficiente? Necessita de um AmpOp pra trabalhar com 5V?
Obrigado!

Gustavomurta, 2020-04-24:
Faço parte da equipe Ambu Sol e Ar, que esta fazendo testes com o sensor acima - NP400 modificado. 
NP-400 - modificado
INPUT - 0 - 60 cmH2O
OUTPUT - 4 - 20mA

Um colega nosso, o Felipe Jonathan, fez algumas medições, usando o circuito que desenhei para ele, seguindo as orientações do engenheiro do fabricante. Como é um sensor analógico industrial, usamos um resistor em série para medir a tensão. Existem módulos específicos para fazer isso, mas não temos em mãos. 
https://www.itead.cc/wiki/4-20mA_Sensor_Brick
Não fizemos aferições de medições ainda, para comprova-las. 

Esse é o diagrama:
[image]

Gustavomurta, 2020-04-28:
Pesquisando documentos da NXP, encontrei mais um sensor muito interessante!  MPXV5004
https://www.nxp.com/docs/en/data-sheet/MPXV5004G.pdf

-  0 to 3.92 kPa, Differential and Gauge (39,97 cm H2O) 

- 1.5% maximum error for 0 to 100 mm H2O

- Alimentação  5,0 V

- Sensitivity - 1 V / KPa  (10,197 cm H2O)

Gustavomurta, 2020-04-28:
**Como calibrar um sensor de 40 cm H20.** 
Eu não sabia que era tão simples!

A melhor calibração deve ser feita em duas partes. Uma pressão deverá ser  medida a 0 mm de H2O e a outra  quando a tubulação estiver a 40 cm de H2O. A diferença nas contagens de ADC divididas por 40 cm fornecerá o número de contagens de ADC (tensão)  por cm. 

Referência : Altitude Pressure EXperimental (APEX) Sensor Board
[https://www.nxp.com/docs/en/application-note/AN3956.pdf](url)

[image]

Samu-vaz, 2020-05-21:
Senhores,
Tenho trabalhado com os sensores da série MPX5700, e está funcionando bem!
Ele faz o trabalho de pressão diferencial e, através de um BMP(E)280, monitoro as alterações de pressão Atmosférica no ambiente, e com um algoritmo bastante simples eh possível fazer a compensação no sistema através dos dados de ambos sensores.
O BME280 nos ajuda ainda na questão de umidade relativa...
Estou trabalhando sensores em redundância:
- Um BMP280 com o BME280 para a variável Pressão Atmosférica;
- Um HTU21D com o BME280 para a variável Umidade Relativa.
Nos 3 sensores monitoro a temperatura também.
Estou bastante satisfeito com o resultado. Caso alguém esteja trabalhando de maneira similar, seria ótimo falar a respeito, trocar experiências e até mesmo aprimorar esse sistema.