Inspire-Poli-USP/Inspire-OpenLung #25: Arruela de teflon

OttoHeringer, 2020-04-03:
Olá pessoal, no projeto (estou analisando a V5) tem uma peça que parece uma arruela de acrílico, que serve de guia para a peça móvel. Na pasta de cortes 2D não tem o DXF dela, aí vem a pergunta. Ela realmente é uma peça cortada em acrílico e esqueceram de gerar o DXF dela ou ela é outra peça e como está tudo a mesma cor eu me confundi? Obrigado.

_Originally posted by @andreribeiroeco in https://github.com/Inspire-Poli-USP/Inspire-OpenLung/issues/15#issuecomment-608226320_

OttoHeringer, 2020-04-03:
Boa observação.
Essa peça é uma "arruela de teflon". Fizemos ela pegando um tarugo cilíndrico de 10 mm de diâmetro e cortando fatias de 5mm numa serra circular. Essas fatias foram colocadas na cortadora a laser e cortadas no tamanho certo.
Usar uma arruela de teflon foi uma dica do HPparanhos que recebemos já faz mais de uma semana, então créditos a ele!

OttoHeringer, 2020-04-03:
Vamos atualizar e colocar esse arquivo faltante.

andreribeiroeco, 2020-04-03:
> Vamos atualizar e colocar esse arquivo faltante.

Boa, aí eu já pego o DXF pra cortar aqui. Obrigado.

closed 2020-04-26 by breno-helf