Inspire-Poli-USP/Inspire-OpenLung #26: Algoritmo de Controle por PID

OttoHeringer, 2020-04-03:
PID servirá para controle da curva de respiração. Mas existem algumas dificuldades em atingir essa curva usando sistema pid, pois após "overshoot" ele continuará fazendo auto compensação com leitura dos sensores, o que pode trazer valores diferentes do esperado. Estamos em busca também de outro tipo de controle para que essa curva seja respeitada. 

Nosso foco está na validação de IHM e controle das válvulas no momento, nosso FORK já está atualizado com ultimas modificações.

_Originally posted by @HPparanhos in https://github.com/Inspire-Poli-USP/Inspire-OpenLung/issues/15#issuecomment-607877154_

OttoHeringer, 2020-04-03:
Ainda na tentativa de abordagem por PID, estamos testando o código [desse repositório](https://github.com/eduoda/odamv).

alberiolima, 2020-04-03:
> PID servirá para controle da curva de respiração. Mas existem algumas dificuldades em atingir essa curva usando sistema pid, pois após "overshoot" ele continuará fazendo auto compensação com leitura dos sensores, o que pode trazer valores diferentes do esperado. Estamos em busca também de outro tipo de controle para que essa curva seja respeitada.
> 
> Nosso foco está na validação de IHM e controle das válvulas no momento, nosso FORK já está atualizado com ultimas modificações.
> 
> _Originally posted by @HPparanhos in [#15 (comment)](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/issues/15#issuecomment-607877154)_

@OttoHeringer, PID para controlar a pressão através de uma válvula, ou o volume?
Quantas e quais são as valvulas que precisam ser controladas?

marcusvscosta, 2020-04-04:
Olá, conheci o seu projeto pela midia e pela internet. Sou professor de controle e automação e trabalho em desenvolvimento e estudo de estratégias de controle. Poderia contar mais em como é modelado o sistema do controle do respirador? Modelo matemático ou métodos de identificação? Tem os pontos de dados para fazer identificação do modelo?  Desde já agradeço a atenção!

OttoHeringer, 2020-04-06:
Oi @alberiolima, eu criei [uma documentação](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/wiki/Entenda-o-Problema) que talvez explique melhor essa questão.

@marcusvscosta obrigado por se disponibilizar.
O andamento do algoritmo de controle está desatualizado aqui no repositório. Isso será atualizado em breve.
Mas eu já posso adiantar por aqui o que está rolando:
- O controle PID não funciona bem para esse problema. Erramos em considerá-lo em primeiro lugar como estratégia de controle.
- Estamos usando uma outra abordagem muito interessante, que é modelar o sistema como um tradicional circuito oscilatório RC, em que a resistência é a resistividade das vias aéreas e a "capacitância" é a chamada complacência do pulmão. A "corrente" é o fluxo de ar. A estratégia é "medir" R e C durante os ciclos de respiração e inspiração para ajustar automaticamente a equação de fluxo de ar para os níveis desejados pelo operador.
  - Uma outra coisa legal disso é que vamos poder fornecer ao operador valores de resistividade das vias aéreas e a complacência do pulmão, que são muito importantes para fazer diagnósticos da situação do paciente "ao vivo".

OttoHeringer, 2020-04-06:
Esse final de semana parece que o algoritmo ando MUITO bem. Vou ver se consigo fazer o Felipe Fava, que é quem está programando isso, pode subir o código aqui no GitHub. Ele está fazendo no MatLab.

marcusvscosta, 2020-04-06:
Se modelar como um sistema RC, um PI já resolve, pois o PID o termo derivativo torna o sistema mais lento, dificultando a chegada ao setpoint de referencia. Trabalho com matlab. Precisa levar também o problema do tempo de amostragem na implementação. Uma forma usual de dimensionar o tempo amostragem é considerar como ao menos 1/10 da constante de tempo dominante do sistema. Tempo de amostragem muito pq gera aliasing (aparecimento de ruidos indesejáveis) e tempo de amostragem grande desobedece o teorema de Nyquist.

alberiolima, 2020-04-06:
> Esse final de semana parece que o algoritmo ando MUITO bem. Vou ver se consigo fazer o Felipe Fava, que é quem está programando isso, pode subir o código aqui no GitHub. Ele está fazendo no MatLab.

OK, eu acredito que possa ajudar com o código! Já que conheço tanto mecânica, quanto programação!

OttoHeringer, 2020-04-06:
Pessoal, criamos outro repositório com o código. Ainda preciso documentar, mas quem entender de controle e saber que estamos usando um modelo de circuito RC para o sistema talvez consiga entender.

Junto com o código, tem uma pasta com todos os dados de caracterização do sistema usando vários testes que viemos fazendo.

https://github.com/Inspire-Poli-USP/Inspire-OpenLung-RCModelFirmware

Quando documentar, coloco aqui.

Esse novo repositório virará um submódulo desde.

alberiolima, 2020-04-06:
Eu vejo um sistema com 3 partes, IHM, Operacional, Sensores/Alarmes. 
Sensores e Alarmes assim como o Operacional são tarefas críticas e acredito que deve ser feitas em processadores diferentes, um conversando com o outro. 
Sensores -> IHM, para mostrar valores/alarme na tela
Sensors -> Operacional, para passar do modo mandatório para assistido (por interrupção)
IHM -> Operacional, para configurações e iniciar/parar
(Sensores também seria capaz de detectar a operação, saber em que ponto está da respiração, e monitorar seu funcionamento correto, e passaria a informação para o IHM) 
Opcionalmente, quando em operação não teria comunicação partindo do Operacional

d-oliveira, 2020-04-07:
> Pessoal, criamos outro repositório com o código. Ainda preciso documentar, mas quem entender de controle e saber que estamos usando um modelo de circuito RC para o sistema talvez consiga entender. [...]

O link está correto?

OttoHeringer, 2020-04-07:
@d-oliveira , tivemos problemas envolvendo a política de disponibilização desse código. Estou pensando como resolver isso.  Se você entrar no [grupo do Telegram](https://t.me/openlungpoliusp) e me procurar por lá, posso falar pessoalmente com vc sobre isso.

Desculpa aí pessoal, resolvendo uns problemas aqui!

sandrompassos, 2020-04-07:
Olá, Pessoal... Sou do aluno de Engenharia de Controle e Automação do IFMS campus Três Lagoas... Estou montando com professores um grupo de estudos na área de Engenharia Clínica... Estamos acompanhando esse projeto e devemos montar um protótipo para testes... Esperamos poder colaborar de alguma forma... Parabéns pela iniciativa...

OttoHeringer, 2020-04-08:
Oi @sandrompassos , me manda uma mensagem no telegram!? Talvez possamos ajudar mandando um protótipo pronto pra vcs irem testando; ainda estamos decidindo para quem vamos mandar uns protótipos, mas um critério é mandar para grupos de ensino e pesquisa.

OttoHeringer, 2020-04-08:
Pessoal, estava conversando com o @emersonmoretto e acho que seria legal organizar as informações sobre os algoritmos de controle que já são usados nos respiradores profissionais; quem puder ajudar, por favor dê uma olhada num novo Issue sobre isso.

juliosouzabrsp, 2020-04-08:
Boa tarde pessoal, sou o Julio Cesar e estive interagindo no grupo do Telegram. O algoritmo de controle dos fabricantes envolvem muitos detalhes e blocos relativos ao controle em si e outros dedicados as condições de contorno ou para acionar alarmes da maneira mais inteligente possível digamos e dentro dos critérios estabelecidos nas normativas vigentes

juliosouzabrsp, 2020-04-08:
Uma prática que vi funcionar nesses casos é instrumentar o protótipo com o máximo de sensores possíveis e ler simultaneamente todos os sinais nos diferentes pontos do sistema, e subsistemas. Estando com os sensores indicando valores razoáveis em todos os pontos, fica mais fácil realizar observações que juntamente com a modelagem teórica devem indicar os parâmetros de controle necessários para estabilidade mínima necessária para segurança do paciente.

juliosouzabrsp, 2020-04-08:
O compartimento interno do sistema de ventilação é dividido em duas partes. O compartimento de entrada ou admissão de oxigênio, que no início do ciclo absorve o gás do cilindro de O2, até que o valor de pressão interna do compartimento seja suficiente para fechar a válvula de admissão e abrir o compartimento de saída/ linha do paciente. Ao transmitir a pressão do O2 para o pulmão do Paciente, se faz o uso de um ciclador controlado, desta forma, o mecanismo faz aumentar a pressão na linha do paciente até o necessário para preencher o pulmão com O2 via traquéias de borracha. Quando o pulmão recebe o volume de ar necessário e pressão, o ciclador fecha o compartimento de saída e o pulmão + caixa torácica, expulsam o CO2 para o ambiente. Nesse momento é preciso ter controle da pressão de saída de CO2 do pulmão para que não fique abaixo da pressão natural do corpo, que mantém o pulmão inflado. Essa é a pressão chamada PEEP.

juliosouzabrsp, 2020-04-08:
Cada compartimento vai sofrer influências que podem ser medidas ou simuladas por blocos para identificação das variáveis mais influentes e refinamento dos controles

juliosouzabrsp, 2020-04-08:
Os modos de falha de qualquer uma das malha de controle e do sistema não podem resultar em situação que possa causar dano parcial ou permanente ao paciente.

juliosouzabrsp, 2020-04-08:
Os controles e atuadores precisam estar configurados de maneira que em caso de falha de algum componente crítico o sistema seja capaz de interromper o funcionamento e acionar um alarme ou entrar em mode de auto recuperação, para garantir a segurança básica e desempenho essencial deste equipamento de suporte a vida

alberiolima, 2020-04-08:
[image]

breno-helf, 2020-04-26:
Esse issue pode ser re-aberto em momento oportuno.

closed 2020-04-26 by breno-helf