Inspire-Poli-USP/Inspire-OpenLung #27: Vídeo e divulgação

juloko, 2020-04-04:
Primeiramente parabéns pelo projeto.

No entanto está faltando vídeo e divulgação do projeto. O projeto alcançará um determinado nível de sucesso quando pessoas leigas na área conhecerem ele e se interessarem, motivará os engenheiros e técnicos ao redor para produzirem...

Um vídeo bobo para rolar no whatsapp já serve (minha visão).

breno-helf, 2020-04-04:
Acho também que outras coisas na parte de divulgação poderiam ser feitas, como criar uma página web (poderia ser no proprio github mesmo, com o github pages) colocando milestones do projeto com fotos e vídeos (com os dados ficaria bem tranquilo organizar isso).

breno-helf, 2020-04-15:
Oba, sei que vocês já apareceram na mídia e tudo, mas ainda existe interesse nisso? Poderia ajudar também. Acredito que se não houver interesse melhor fechar a issue.

breno-helf, 2020-04-20:
Oi @juloko !

Acabei de achar um vídeo bacana que poderia rodar no whatsapp, como você sugeriu. https://youtu.be/hkaXHtFTmG4 , isso aqui serve?

Nesse caso vou fechar essa Issue e abrir outra mais relativa a uma página de divulgação que faremos mais adiante.

juloko, 2020-04-22:
Perfect! Parabéns! Vou baixar e divulgar.

juloko, 2020-04-23:
@breno-helf gostaria de contribuir com a página de divulgação.

breno-helf, 2020-04-23:
Tranquilo @juloko , estou pensando em abrir um github pages daqui uns dias, te mantenho atualizado!

breno-helf, 2020-04-30:
Vou reabrir esse Issue pois acredito que podemos fazer um github pages legal com a documentação que temos no repositório e na wiki.