Inspire-Poli-USP/Inspire-OpenLung #29: Criar um sistema de Integração contínua

breno-helf, 2020-04-04:
Acredito que isso seja interessante, mesmo o projeto sendo focado em hardware podemos configurar um sistema de integração contínua para assegurar certas qualidades nas contribuições. Poderíamos configurar um sistema como https://travis-ci.com/ para "testar" contribuições como:
 - Verificar se dados novos são submetidos no formato correto
 - Verificar se mudanças de software passam testes existentes
 - Verificar se mudanças no relatório não apagam coisas que não deveriam ser apagadas

Isso seria um passo em assegurar a qualidade de novas contribuições.

TiagoDanin, 2020-04-30:
@OttoHeringer Estou trabalhando nisto!

breno-helf, 2020-04-30:
Obrigado @TiagoDanin , já vou dar uma olhada.

PR #68 (2020-04-30): Teste de documentação com CI
### Questões básicas

* [x] Eu li o [Código de conduta](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/blob/master/CODE_OF_CONDUCT.md) e as [Guidelines de contribuição](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/blob/master/CONTRIBUTING.md) antes de submeter o Pull Request
* [x] Eu chequei antes se não habvia nenhum [Pull Request](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/pulls) resolvendo o mesmo problema.

### Mudanças na documentação

Close #29

- Travis CI integração
- Usar Remake Lint para testa documentação
- Corrigir falhas apontad [... cut]
closed 2020-04-30 by breno-helf