Inspire-Poli-USP/Inspire-OpenLung #31: Papers e materiais didáticos

VitorFrost, 2020-04-06:
Pessoal, podemos listar aqui as principais referências que estão usando? 
Tentar indicar o assunto das referências ao postar aqui.
Obrigado.

OttoHeringer, 2020-04-06:
https://www.youtube.com/watch?v=7vLPefHYWpY

OttoHeringer, 2020-04-06:
Essa aqui é mais um texto crítico e não tão explicativo quanto o vídeo que linkei antes
https://hackaday.com/2020/04/05/making-an-arduino-ventilator-read-this-first/

sandrompassos, 2020-04-08:
**Recomendações Brasileiras de Ventilação Mecânica**

http://www.scielo.br/pdf/rbti/v26n2/0103-507X-rbti-26-02-0089.pdf
http://www.scielo.br/pdf/rbti/v26n3/0103-507X-rbti-26-03-0215.pdf

**Diretrizes Brasileiras de Ventilação Mecânica**

https://interfisio.com.br/imagens/artigos/2013/Diretrizes-AVM-AMIB-SBPT-2013.pdf

danilochui, 2020-04-10:
Artigo sobre modelo do sistema respiratório:

[COB03-1961.pdf](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4462036/COB03-1961.pdf)