Inspire-Poli-USP/Inspire-OpenLung #32: Esquematização do Sistema Completo

OttoHeringer, 2020-04-07:
Eu vejo um sistema com 3 partes, IHM, Operacional, Sensores/Alarmes. 
Sensores e Alarmes assim como o Operacional são tarefas críticas e acredito que deve ser feitas em processadores diferentes, um conversando com o outro. 
Sensores -> IHM, para mostrar valores/alarme na tela
Sensors -> Operacional, para passar do modo mandatório para assistido (por interrupção)
IHM -> Operacional, para configurações e iniciar/parar
(Sensores também seria capaz de detectar a operação, saber em que ponto está da respiração, e monitorar seu funcionamento correto, e passaria a informação para o IHM) 
Opcionalmente, quando em operação não teria comunicação partindo do Operacional

_Originally posted by @alberiolima in https://github.com/Inspire-Poli-USP/Inspire-OpenLung/issues/26#issuecomment-610095253_

OttoHeringer, 2020-04-07:
Obrigado por sumarizar esses pontos, @alberiolima.
Acho importante fazer isso para também apontar áreas onde há menos esforços ainda.

Você poderia fazer um diagrama sobre isso? Muita gente está pedindo algo do tipo para entender melhor o contexto de onde o projeto sendo documentado nesse repositório se encaixa.

alberiolima, 2020-04-08:
Fiz um rascunho rápido! De como eu acredito que seria a base inicial. 
[image]

alberiolima, 2020-04-08:
[image]

AlvaroAguiar, 2020-04-14:
Teremos uma visão explodida do projeto? Isso facilita para o leigo que porventura queira produzir o equipamento (meu caso). Cada parte com o nome da peça e o link para o corte a laser (sugestão).

breno-helf, 2020-04-15:
Oi @alberiolima , você se importaria de submeter um Pull request com essas esquematizações? Podemos iterar sobre elas e tornar em uma contribuição técnica para o projeto!

Obrigado!

alberiolima, 2020-04-16:
> Oi @alberiolima , você se importaria de submeter um Pull request com essas esquematizações? Podemos iterar sobre elas e tornar em uma contribuição técnica para o projeto!
> 
> Obrigado!

Olá @breno-helf , não entendi, o que exatamente quer que faça? Qualquer coisa me envia um e-mail, tem no meu perfil do github. Abraço

breno-helf, 2020-04-20:
Oi @alberiolima !

Vou mandar um email com essa mesma mensagem, mas acredito que seu esquema que você compartilhou nessa Issue poderia entrar no repositório do projeto. Para isso seria nescessário um Pull Request para incluir elas. Tem uma explicação no repositório de como fazer isso (https://github.com/Inspire-Poli-USP/Inspire-OpenLung/blob/master/CONTRIBUTING.md#como-fazer-as-contribui%C3%A7%C3%B5es). Colocando esse seu esquema como contribuição, além de você entrar como um dos contribuidores do projeto, sua contribuição passa a ser uma contribuição técnica que vai ser usada por outras pessoas desenvolvendo o respirador!

Qualquer dúvida sobre como fazer um pull request, ou caso você não possa fazer, me avise! Eu posso fazer o Pull Request por você se você quiser (Mas nesse caso infelizmente seu nome não fica como de contribuidor). 

Abraços!