Inspire-Poli-USP/Inspire-OpenLung #34: Dúvidas sobre o controle da ventilação nos pulmões

MarioCarrazza, 2020-04-08:
Pelo que observei o equipamento que faz a função do pulmão exibe no display a curva de ventilação. Entendo que este display também deva fazer parte do projeto para que o equipamento ventilador possa regular a velocidade do motor para que permaneça dentro das necessidade de ventilação.

alberiolima, 2020-04-09:
> Pelo que observei o equipamento que faz a função do pulmão exibe no display a curva de ventilação. Entendo que este display também deva fazer parte do projeto para que o equipamento ventilador possa regular a velocidade do motor para que permaneça dentro das necessidade de ventilação.

Não amigo,um display é apenas um dispositvo que mostra ao humano dados que estão sendo processados dentro da máquina, faz parte do que chamados de IHM (interface home-máquina), um display é apenas um monitor, tipo um monitor de computador, só serve para você olhar para ele.

MarioCarrazza, 2020-04-09:
Entendo que as informações que são mostradas nesse display seriam necessárias para efetuar um ajuste no ventilador caso a curva de ventilação esteja fora da especificação.

alberiolima, 2020-04-09:
> Entendo que as informações que são mostradas nesse display seriam necessárias para efetuar um ajuste no ventilador caso a curva de ventilação esteja fora da especificação.

A curva é apenas a representação dos dados lidos, ela não é uma especificação.
Especificação seria: volume,tempo,pressão,etc!
Alguém me corrija se eu estiver errado!