Inspire-Poli-USP/Inspire-OpenLung #35: Simulador do Respiraor Dixtal DX 3010

sandrompassos, 2020-04-08:
Para ilustrar o funcionamento de um respirador comercial, seguem abaixo, links de um simulador do Respirador Dixtal DX 3010 e de dois vídeos com a demonstração da montagem e inicialização do respirador real:

Montagem e inicialização Dixtal DX 3010
https://youtu.be/SYe3fqq6nTg
https://youtu.be/XWzXESjcVJI

Simulador Dixtal DX 3010 
http://www.4shared.com/rar/sQECOE0jce/SIMULADOR_DIXTAL.html

OttoHeringer, 2020-04-11:
Oi Sandro, muito boa essa sugestão de simulador. Resolve um grande gargalo da contribuição da comunidade, que é a falta de acesso a equipamentos de simulação.

Depois que alguém tentar usá-lo e colocar a experiência aqui, quero colocar a indicação de uso no README, no CONTRIBUTING e/ou criar uma página da wiki documentando como usá-lo.

m-rodrigues-souza, 2020-04-13:
Bom dia,

Segue mais um simulador:

https://girardi.blumenau.ufsc.br/sdvm/

O pessoal da UNIMAR que nos indicou.

OttoHeringer, 2020-04-14:
@sandrompassos, você consegue transformar esse apanhado de simuladores e roteiros de utilização numa página da wiki e mandar um pull-request? Não precisa se preocupar muito com o formato que eu posso editar depois.

sandrompassos, 2020-04-15:
Posso sim... Só preciso de mais um pouco de tempo para me familiarizar melhor com essas meios digitais de divulgação acadêmica... Aos poucos vou preparando o material... 

Achei muito interessante esse simulador disponibilizado pelo @m-rodrigues-souza... Essa apresentação dos gráficos de pressão, volume e vazão dão uma excelente ideia do funcionamento de um respirador...

RBMarvila, 2021-04-14:
Boa tarde,

Os simuladores são ótimos, Sandro ou algum colega teriam o código de acesso desses aparelhos para zerar as 500H?

ANAPAULAFISIO, 2022-01-06:
Olá! Alguém tem o link para baixar o simulador da INTERMED ??????

katiadouglebeniamatos-del, 2026-10-15:
como baixo o simulador no notebook

sandrompassos, 2026-10-15:
Boa tarde...
Infelizmente, os links acima não estão mais disponíveis... Segue um novo link com o arquivo:

https://drive.google.com/file/d/1GpCO7pf3O_X_YKcP1d_STWTtb3bel8ZC/view?usp=drive_link

Não há necessidade de instalação... Basta baixar e executar o arquivo...
Sandro M. Passos