Inspire-Poli-USP/Inspire-OpenLung #36: Sensor de Fluxo para Controle de Volume

m-rodrigues-souza, 2020-04-08:
Olá a todos.

@OttoHeringer e @emersonmoretto.

Vocês têm dados atualizados ou novidades sobre o controle de volume? Vimos que fizeram uma curva interpolada da resposta do ambu. Isso está validado? É confiável? Dessa maneira todo ambu deve ser calibrado?

E caso já tenham tido contato com sensor de fluxo, saberiam indicar algum part number?
Encontramos um da Honeywell (HAFUHT0100L4AXT), porém, como já era esperado, é caro (168 USD).

Obrigado!

alberiolima, 2020-04-09:
Acredito que não seja complicado construir (desde que tenha como calibrar). Usando um sensor diferencial de pressão, e o legal é que parte do componente já exite como acessório de ventiladores. 
[image]
O que precisa saber é a área dos furos, se alguém tiver a folha de dados, ou mesmo puder fazer leitura e comparar com um sensor de fluxo real, acredito que não seja complexo conseguir as constantes para a formula do calculo do fluxo.

alberiolima, 2020-04-09:
Mas para usar o "Ambu", talvez um sensor de fluxo seria necessário apenas para saber se tem algum vazamento no sistema. Pois o fluxo é a quantidade de ar que se envia pelo tempo.
Se o volume conseguido com o  "Ambu" for 500ml, e o dispositvo conseguir enviar isso em 1s, a vazão é 0,5L/s. Acredito que a parte mai complexa do sistema usando "Ambu", é como coneguir essa velocidade, empurar 500ml de ar em 1s. Qual motor teria essa velocidade. Se precisa de um fluxo de 250ml/s então só precisa ir até a metade do curso, em um segundo, ou seja metade da velocidade!

buenow, 2020-04-10:
@alberiolima O SFM3300 está em falta até na mouser.

buenow, 2020-04-10:
@alberiolima Eu não entendi o motivo de não termos caminhado em direção a motores em vez de ambu. Algo como esse motor...https://www.aliexpress.com/i/32627605609.html. Algum motivo espeífico?

alberiolima, 2020-04-10:
> @alberiolima Eu não entendi o motivo de não termos caminhado em direção a motores em vez de ambu. Algo como esse motor...https://www.aliexpress.com/i/32627605609.html. Algum motivo espeífico?

O Ambu já é um equipamento testado e homologado, tem todos os conectores, filtros, adaptadores, sensores,etc.. criar tudo do zero será possível, só que bem mais complexo

Gustavomurta, 2020-04-18:
Alguém já pensou em usar um sensor MAP ou sensor MAF ? Similar ao usado em motor de veículos? É claro, o fluxo de ar de entrada do motor é bem maior. Estou pesquisando esse tipo de sensor. 
https://en.wikipedia.org/wiki/MAP_sensor
https://en.wikipedia.org/wiki/Mass_flow_sensor

Gustavo Murta 
Ambu Sol e Ar

alberiolima, 2020-04-18:
> Alguém já pensou em usar um sensor MAP ou sensor MAF ? Similar ao usado em motor de veículos? É claro, o fluxo de ar de entrada do motor é bem maior. Estou pesquisando esse tipo de sensor. [...]

Encontrei soluções que pareciam interessantes, pesquisando por "Spirometer"

paulocg, 2020-04-22:
> Alguém já pensou em usar um sensor MAP ou sensor MAF ? Similar ao usado em motor de veículos? É claro, o fluxo de ar de entrada do motor é bem maior. Estou pesquisando esse tipo de sensor. [...]

Gustavo em um grupo do Telegram [1]  específico sobre desenvolvimento de um sensor de fluxo de ar para substituir o sensor SFM3300 da Sensirion utilizado no projeto do ventilador REESistencia Team eu encontrei a indicação deste projeto - Airflow Sensor - Automotive Sensor AFH55M12 [2].

[1] CV19ElecCaud0321 - Solicitar inclusão no grupo do Telegram -  Ignacio Arevalo (@nachogazpacho)
[2] https://github.com/hydronics2/COVID-19-Airflow-Sensor-AFH55M12

Paulo Gonçalves
UTFPR

Gustavomurta, 2020-04-22:
Paulo Gonçalves, Obrigado pela indicação! 
Os sensores MAF de carro são projetados para volumes de ar bem maiores do que numa respiração.Não sabia que podem ser usados em Respiradores. 
Obrigado 
Gustavo Murta 
Ambu Sol e Ar

JonathasKerber, 2020-04-23:
> Acredito que não seja complicado construir (desde que tenha como calibrar). Usando um sensor diferencial de pressão, e o legal é que parte do componente já exite como acessório de ventiladores. [...]

Alguma atualização sobre essas medidas? Também não encontro a folha de dados.