Inspire-Poli-USP/Inspire-OpenLung #42: Volume do Ambu

lfelipev, 2020-04-11:
Boa noite a todos!

Gostaria de saber qual o volume do Ambu da versão atual do projeto, é uma dúvida que tenho sobre o volume do Ambu ser suficiente para manter uma pressão constante no intervalo de tempo T2 da curva característica.

alberiolima, 2020-04-11:
> Boa noite a todos!
> 
> Gostaria de saber qual o volume do Ambu da versão atual do projeto, é uma dúvida que tenho sobre o volume do Ambu ser suficiente para manter uma pressão constante no intervalo de tempo T2 da curva característica.

A curva é apenas o gráfico do comportamento do sistema respiratório, a expiração ela acontece naturalmente, a pressão vai diminuindo porque o ar vai saindo do sistema, você precisa apenas fechar a válvula quando chegar na pressão PEEP. Se o dispositivo já tiver uma válvula de PEEP, você precisa apenas aguardar o final do tempo T2 para iniciar uma nova inspiração.
_(informação confirmada com médico)_

alberiolima, 2020-04-11:
> Boa noite a todos!
> 
> Gostaria de saber qual o volume do Ambu da versão atual do projeto, é uma dúvida que tenho sobre o volume do Ambu ser suficiente para manter uma pressão constante no intervalo de tempo T2 da curva característica.
Observando o grafico, o fluxo de ar é constante em INSP (T1), sai de zero vai direto para o valor expecífico, a pressão em T1, representa a resistência que o sistema repiratório do paciente oferece ao fluxo. Volume representa a quantidade de ar que entrou. Em T2, pressão representa a força que o sistema respiratório faz para empurrar o ar para fora.

[image]

OttoHeringer, 2020-04-14:
Oi @lfelipev,

boa pergunta, isso não está especificado no nosso BOM na página inicial. Vou colocar lá.

Usamos um Ambu com capacidade de 1600 mL. E sim, estamos conseguindo "entregar" o necessário.

@alberiolima , excelente comentário, vou referenciá-lo na documentação do "Entendendo o Problema".

OttoHeringer, 2020-04-14:
Atualização da documentação sobre respiração linkando esse Issue feita aqui:
https://github.com/Inspire-Poli-USP/Inspire-OpenLung/wiki/Entenda-o-Problema#vari%C3%A1veis-de-cada-paciente
Lista de materiais (BOM) atualizada aqui:
https://github.com/Inspire-Poli-USP/Inspire-OpenLung#partes-m%C3%A9dicas

closed 2020-04-14 by OttoHeringer