Inspire-Poli-USP/Inspire-OpenLung #43: Desafio: Válvula PEEP

OttoHeringer, 2020-04-11:
Oi pessoal!

Temos um desafio para quem quer ajudar com soluções de hardware: desenvolver uma válvula PEEP.

Informações específicas sobre isso estão documentadas [nessa página da wiki](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/wiki/Desafio:-V%C3%A1lvula-PEEP).

Conforme descrito no link anterior, vamos discutir e compartilhar conteúdos por aqui!

danilochui, 2020-04-11:
Boa noite. Vi que na página da descrição do desafio, é sugerido que se use sensores automotivos. Fazendo uma pesquisa rápida aqui, me deparei com esse vídeo no Youtube. O rapaz lê um sensor MAP automotivo com Arduíno. Aqui: https://www.youtube.com/watch?v=83LuzJTIbAw
Dá para ver que ele mede a pressão ambiente como 1000,XX mBar. Os valores depois da vírgula já possuem uma incerteza de uns 0,5 mBar. Como 1 cmH2O é 0,98 mBar, então temos uma incerteza de aproximadamente 0,5 cmH2O. 
Lá na página fala: "Ter precisão nos valores configurados - ou seja, por exemplo: selecionar uma pressão de 7 cmH2O deve manter uma medida de pressão real de 7 cmH20"
Na verdade, usando um MAP automotivo, teríamos 7 +- 0,5 cmH2O. Isso é aceitável?

adalmirabdala, 2020-04-13:
Princípio: Uso de VENTILADOR como regulador de PRESSÃO.
Boa noite. Segue uma alternativa de válvula PEEP, evitando: a)mola (fadiga), b)válvulas c)obstrução na ocorrência de falha. Tem controle preciso no PWM do arduino, na escala requerida. Também utilizei o BME280 no arduino para aferir as medidas práticas encontradas.
No protótipo utilizei um ventilador 12vcc de secador de cabelo + "adaptação" para cano com escala, como segue em foto. Esse ventilador não tem a capacidade requerida, mas o princípio foi demonstrado.
Princípio: usar ventilação forçada na saída da traqueia (expiração), no sentido de criar contrapressão constante no sistema, que pode ser vencida pelo paciente sem restrição.
Medição de pressão positiva: 3,4mmH2O (medição utilizando BME280, BMP280 >>955,0-951,7 hPa)
Teste prático: montagem do sistema com saída para baixo em copo de água, observado entre 3 e 3.5 cm o vencimento da coluna de água e formação de bolhas.
Problemas: ruído, a contrapressão soma com a pressão do ambu.
Obs.: O BME280 tem 2 endereços, podendo ter um segundo redundante em outro ponto.
[image]
[image]

adalmirabdala, 2020-04-13:
obs: deve ter mais uma válvula unidirecional para evitar risco de reinalação.

adalmirabdala, 2020-04-13:
Valvula PEEP utilizando a própria coluna dágua como válvula.
Segue abaixo esboço de sistema utilizando a própria coluna dágua como agente pressurizador do sistema de expiração + válvula unidirecional. Ainda não montei um protótipo, porém é o mais simples que consegui.
Desvantagem: regulagem manual e dificuldade de estanqueidade do sistema.
[image]

danilochui, 2020-04-13:
Colocando duas válvulas solenoides uma de entrada de água e outra de saída e uma bóia, é possível fazer essa regulagem manual virar eletrônica.

OttoHeringer, 2020-04-14:
@adalmirabdala suas ideias são muito criativas. A mais interessante das duas é o controle por nível de água, e o @danilochui complementou bem sobre como controlar isso eletronicamente. Essa ideia é elegante porque além de ser um pouco mais simples que a do ventilador, ela "resolve" um problema que tem no ventilador: contaminação. Alguns projetos de ventiladores emergenciais inclusive estão propondo usar essa "armadilha de água" no final com a finalidade principal de ajudar evitar contaminar o ar ambiente com o que passou pelo paciente e vice-versa - porque em tese, boa parte do que viria junto com o ar expirado ficaria na água.

O que estamos procurando é algo mais na linha do que o @danilochui estava pesquisando: um componente pronto que originalmente é usado para uma finalidade mas que resolveria quase que perfeitamente nosso problema.

OttoHeringer, 2020-04-14:
Aproveitando, vejam o que o @emersonmoretto estava trabalhando no final de semana:
https://youtu.be/Lm-YNxPOsxA

É uma proposta simples, mas que ainda está sendo testada aqui. Não queríamos depender dela porque claramente é mais complexa do que pegar um componente "pronto" e adaptá-lo.

Pensando bem aqui, o que precisamos, no fundo, é: uma "torneira" que a gente consiga ficar controlando o quanto abrimos e fechamos através de um arduino, por exemplo.

felipeosi, 2020-04-14:
Tem uma solução comercial para controle de boost em carros turbo de corrida.  Usa bico injetor de carro, que suporta infinitos (quase) ciclos de fechamento e abertura, e o tempo morto de abertura tá na ordem de 0,80 milisegundos pra 12 volts no acionamento.

https://produto.mercadolivre.com.br/MLB-1220289052-suporte-bicos-para-booster-eletronico-integrado-injeco-_JM?matt_tool=86155663&matt_word&gclid=CjwKCAjwvtX0BRAFEiwAGWJyZJ_YDjBeiw1HFAIqMFFgC_jmCoKLNmMKpJqoB8pvnZefIgZnXFpCVBoCrXgQAvD_BwE&quantity=1

1 bico controla o envio do gas pressurizado para o volume de controle, e o outro a liberação do gás para atmosfera. No caso do ventilador, talvez usar só um bico injetor liberando o gás do volume de controle para atmosfera seja suficiente. Mas se não, tem a opção de colocar o outro bico captando ar do zona pressurizada do ambu, e enviando para o volume de controle.

emersonmoretto, 2020-04-15:
> oço de sistema utilizando a própria coluna dágua como agente pressurizador do sistema de expiração + válvula unidirecional. Ainda não montei um protótipo, porém é o mais simples que consegui.

é então, esse era o jeito que usavam na década de 80 né. Mas parece que não se pode mais usar desse jeito por questões de contaminação e tem um problema de excesso de bolhas que faz com que ela não funcione bem (quando em peep alto, solta tanta bolha que afeta o mecanismo de pressao da água e nessa situação ele falha e acaba segurando pra baixo o valor)

emersonmoretto, 2020-04-15:
> a: "Ter precisão nos valores configurados - ou seja, por exemplo: selecionar uma pressão de 7 cmH2O deve manter uma medida de pressão real de 7 c

Olha, pra fins do algoritmo de controle essa resolução não ajuda muito, na verdade acaba proporcionando um erro de fitting de curva considerável. A gente tá usando um sensor com resolução de 44 mV/cm H2O pra ficar legal. ex: mpx5010

emersonmoretto, 2020-04-15:
> Tem uma solução comercial para controle de boost em carros turbo de corrida. Usa bico injetor de carro, que suporta infinitos (quase) ciclos de fechamento e abertura, e o tempo morto de abertura tá [...]

Infelizmente bico injetor automotivo não dá... 1) ele tem resistencia muito grande (acima de 15), essa resistencia causaria um enorme atraso na saída fazendo o paciente não exalar o ar totalmente, o que por consequencia, aumentaria MUITO o CO2 no corpo, 2 minutos com vazão prejudicada o paciente já pode ter dano; e 2) vazão, um bico injetor tem uma vazão de alguns mL/min só, a expiração chega a ter 600ml em pouco mais de 1 segundo

emersonmoretto, 2020-04-15:
> Princípio: Uso de VENTILADOR como regulador de PRESSÃO.
> Boa noite. Segue uma alternativa de válvula PEEP, evitando: a)mola (fadiga), b)válvulas c)obstrução na ocorrência de falha. Tem controle [...]

Muito criativa a ideia, mas acho que isso não segura em situações extremas como em 15 cm h2o.
Teria que ser um sistema muito vedado e um ventilador muito forte.

VitorFrost, 2020-04-15:
O maior problema dos ventiladores é conseguir o produto rápido. Poderia utilizar-se de dois ventiladores em série, para uma pressão maior. 
Só que isso vai gerar um barulho no estilo de aspirador de pó, vai atrapalhar muito em ambiente hospitalar e elevar o stress de todos. Não podemos deixar isso de lado.

VitorFrost, 2020-04-15:
A minha grande dúvida é, vocês sabem aonde está ocorrendo a falha da válvula Peep? É a mola que entra em fadiga? Ou é a borracha de vedação?
Se for a borracha, não é ressecamento da mesma?

VitorFrost, 2020-04-15:
O mais fácil e seguro seria automatizar a válvula Peep mecânica já existente por meio de um servomotor

OttoHeringer, 2020-04-16:
> A minha grande dúvida é, vocês sabem aonde está ocorrendo a falha da válvula Peep? É a mola que entra em fadiga? Ou é a borracha de vedação?
> Se for a borracha, não é ressecamento da mesma?

ATUALIZAÇÃO

Desenvolvemos uma solução seguindo essa ideia que o @VitorFrost deu!

Ainda não ficou perfeita porque o hardware precisa ser melhor; em breve divulgaremos a documentação.

Mais informações na [página atualizada da wiki](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/wiki/Ajuda-da-Comunidade-com-a-V%C3%A1lvula-PEEP) sobre esse asunto.

VitorFrost, 2020-04-16:
@emersonmoretto 
Eu pensei numa melhoria nesse sistema que vocês pensaram. Nesse modo aqui, o ajuste é feito com 1/4 de volta apenas, dando velocidade na resposta, e aumentando as frações de pressão disponíveis.
E eu fiz no papel na verdade metade da válvula, o ideal é ela ser espelhada no eixo, ficando duas janelas opostas e a parte móvel em formato quase de um 8.
[image]
[image]

filzek, 2020-04-16:
A esterilização da saída poderia ser por uma micro tela energizada com íons,  de maneira que passe por uma UV, ozônio, estes micro disparos eletrônicos poderiam formar um esterilizador de passagem,

felipeosi, 2020-04-16:
> A esterilização da saída poderia ser por uma micro tela energizada com íons, de maneira que passe por uma UV, ozônio, estes micro disparos eletrônicos poderiam formar um esterilizador de passagem,

Tem filtros HEPA que já são usados nos aparelhos comerciais e são super baratos. Pode melhorar a filtragem com mais esses recursos, sem dúvida

felipeosi, 2020-04-16:
[https://grabcad.com/library/valvula-peep-pra-respirador-pulmonar-mecanico-1](url)

Esse design vai a valvula de prioridade automotiva e um atuador linear. Usar um solenoide pode ser uma opção, mas não estou seguro quanto a capacidade de vedação com solenoide. Pq precisa ter uma pressão na sede da valvula pra vedar.

[image]

lmsrv, 2020-04-17:
Pesquisando em relação às válvulas PEEP, desde o início achei a ideia da válvula Boehringer a mais fácil de ser replicada e com maior confiabilidade.

Ela funciona baseada no peso de esferas, que são erguidas de acordo com a pressão de expiração do paciente.

Dessa maneira, apresenta uma resposta constante ao longo do tempo e não depende de molas, imãs, motores, etc.

Os maiores inconvenientes são:

•	Ela tem que operar na vertical.
•	O ajuste tem que ser manual.

Desenhei um modelo semelhante, que trabalharia com pequenas “arruelas”. Essas “arruelas” acrescentariam peso conjunto, regulando o sistema em intervalos de 1 cmH2O.

Seria interessante enviar 02 conjuntos com as “arruelas” (peças 02 e 03) com o aparelho. Uma seria estaria sendo utilizada e segunda serviria para fazer os ajustes se for necessário fazer a troca.

Para trocar, basta retirar a tampa e inserir o novo conjunto em questão de segundos.

[image]
[image]
[image]
[image]

Gustavomurta, 2020-04-18:
Boa tarde, estou vendo que estão tentando usar válvulas mecânicas. 
Por que não usar um sensor eletrônico de pressão? Existe alguma limitação em usá-lo?

1) MPX5010 - 0 a 101,97 cmH2O
https://www.nxp.com/docs/en/data-sheet/MPX5010.pdf
2) MPX10 
https://www.nxp.com/docs/en/data-sheet/MPX10.pdf

Gustavo Murta 
Ambu Sol e Ar 
https://www.instagram.com/ambusolear/

VitorFrost, 2020-04-23:
Para automatizar uma válvula Boehringer, podemos trabalhar com imãs, criando uma _mola magnética_ !
A ideia básica é a parte móvel inferior ser construída com um ímã. Existem soluções industriais assim:
[image]

E na parte exterior do copo, aproximamos ou afastamos um outro ímã em repulsão!
Como no desenho que fiz abaixo (obviamente o desenho não está em escala) :
[image]

lmsrv, 2020-04-24:
Olá, boa tarde.

A ideia da válvula magnética é interessante. Existem dessas válvulas para vender no mercado (mas não são automatizadas).

[image]
[image]

O conceito é bem antigo e foi proposto em 1976:

A Magnetic PEEP Valve (no Google é possível achar o estudo)

CHARLES F. HOBELMANN, JR, LCDR (MC) USNR"
DAVID E. SMITH, LCDR (MC) U S N t
RICHARD W. VIRGILIO, CDR (MC) USNS
San Diego, California
Paper received: 9/15/76

[image]
[image]

Daria para automatizar, através de 02 opções:
- Eletroímã (regulando a voltagem e dosando, dessa forma a força e a abertura).
- Aproximação de ímãs, conforme você propôs.

 Mas aí ela fugiria das principais qualidades (no meu ponto de vista) das válvulas Boehringer:
- Extremo baixo custo
- Simplicidade
- Facilidade de fabricação
- Confiabilidade elevada

Mas, com certeza, também é uma opção válida.

Abs

fperez51, 2020-04-24:
Tente usar o conjunto VUP da empresa Lumiar.

putzgrilous, 2020-04-28:
Pelo que entendi, a válvula PEEP poderia ficar numa bancada, certo? Caso afirmativo, proponho o seguinte: um corpo de borboleta eletrônico de qualquer carro com motor moderno 1.0 8V (ou menor) consegue regular até a marcha lenta, tamanha a precisão de posicionamento da válvula. É capaz de regular um grande fluxo de ar e tem resposta quase instantânea.

Sei que é uma peça grande e pesada, mas atende aos requisitos de custo baixo, confiável, fácil de encontrar e eletrônica. Além disso, é esterelizável e funciona em qualquer posição.

Tenho certeza de que usando dois sensores de pressão absoluta disponíveis para Arduino é possível fazer um controle de abrir/fechar a borboleta, mantendo a pressão de ar do circuito constante, independentemente do fluxo de ar regulado. Poderia até fechar a válvula, se quiser, interrompendo o fluxo de retorno e evitando a retro-contaminação.

Acho que a única dificuldade seria fazer um algoritmo de controle rápido e estável, mas essa eu deixo pro pessoal do software... :-)

VitorFrost, 2020-04-30:
> Pelo que entendi, a válvula PEEP poderia ficar numa bancada, certo? Caso afirmativo, proponho o seguinte: um corpo de borboleta eletrônico de qualquer carro com motor moderno 1.0 8V (ou menor) [...]

Discutimos um pouco essa questão no canal do telegram, e a grande difículdade em utilizar está válvula está no oewueno

VitorFrost, 2020-04-30:
@putzgrilous tudo bem?
Discutimos essa opção no grupo do telegram, pois as válvulas borboletas são ótimas para controle de fluxo rapidamente e com razoável precisão. Mas ela apresenta vazamento, o que pode inviabilizar o seu uso. Precisaria medir quanto ar vaza, para poder utilizar ela.
E por não ser um sistema selado, pode ocorrer a entrada de materiais no motor.

putzgrilous, 2020-04-30:
Um corpo de borboleta automotivo é absolutamente estanque, seja na parte elétrica do conjunto eletro mecânico que precisa suportar a lavagem do motor com jato de água, seja na própria válvula de fluxo, que precisa assegurar vácuo completo do motor, quando fechada, para o sistema de freios do carro funcionar.

putzgrilous, 2020-04-30:
O único conhecimento "especial" que você precisa ter sobre esse componente é que, para fechar totalmente, você precisa acionar o motor com tensão inversa, pois na posição normal da válvula (ou não energizada) ela é levemente aberta para manter a marcha-lenta do carro em caso de falha.

VitorFrost, 2020-05-01:
Por algum motivo meu celular fechou a questão! Peço desculpas!

luis-possatti, 2020-05-07:
Pessoal, talvez essa proposta de assemelhe a outras ideias já dadas anteriormente, mas vamos lá:
A solução que eu vejo seria a seguinte:
 um conjunto de duas válvulas pneumáticas de forma que uma controla o fluxo de ar do AMBU para o paciente e outra libera o fluxo de ar do paciente para o ambiente. 

Durante o processo de expiração, bastaria medir a pressão e quando ela chegasse ao valor de PEEP, fechar a válvula que controla a expiração. Dessa maneira o ar iria parar de fluir e manteria a pressão constante em PEEP