Inspire-Poli-USP/Inspire-OpenLung #44: Força axial aplicada no pistão/motor

maximemontoya, 2020-04-11:
Boa noite, 
Gostaria de saber se vocês conseguiram medir a força aplicada pelo ambu sobre o pistão durante aplicação da pressão de pico? 
Analiticamente esta força se encontraria em torno de 24N. 
No modelo 3D e nos vídeos, não consegui ver se existia uma peça impedindo a transmissão desta força, do ambu sobre o pistão, direitamente ao motor via o fuso e acoplador.  
Levando em conta as specs deste tipo de motor, um força axial admissível máxima de 15N (max encontrada na internet), na ausencia desta peça,  a repetição desta força axial não poderia prejudicar o funcionamento do motor ao longo prazo?

OttoHeringer, 2020-04-14:
Oi @maximemontoya, entendi seu questionamento. Realmente não estamos protegendo o motor dessa força.

Se não me engano, o @emersonmoretto tinha mencionado que esse motor tem qualidade suficiente para aguentar para essa aplicação em particular sendo feita e também para a expectativa de uso a longo prazo (meses, não anos). Mas vou deixar aqui para ele responder depois.

Você poderia compartilhar onde achou essas specs?

maximemontoya, 2020-04-14:
segue documentação das specs motor NEMA 23 15kgf 3 A:
https://storage.googleapis.com/baudaeletronicadatasheet/AK23-15F6FN1.8.pdf
https://componentesamericana.com.br/wp-content/uploads/2017/02/MOTOR-DE-PASSO-AK2315F6FN1.8-COMPONENTES-AMERICANA-FONE-19-3012-6360.pdf
https://components101.com/sites/default/files/component_datasheet/NEMA23%20Stepper%20Motor.pdf