Inspire-Poli-USP/Inspire-OpenLung #47: Manual de Montagem

OttoHeringer, 2020-04-14:
Obrigado pela sugestão. O melhor mesmo seria um manual de montagem. Aliás, isso é um ótimo novo Issue.

_Originally posted by @OttoHeringer in https://github.com/Inspire-Poli-USP/Inspire-OpenLung/issues/46#issuecomment-613487881_

Eu vi que o manual do fork do @HPparanhos está bem legal.
https://github.com/HPparanhos/CITI-OpenLung/blob/master/Manual%20de%20montagem/564135%20(1).pdf

Alguém teria uma sugestão a mais?

AlvaroAguiar, 2020-04-14:
> Obrigado pela sugestão. O melhor mesmo seria um manual de montagem. Aliás, isso é um ótimo novo Issue.
> 
> _Originally posted by @OttoHeringer in [#46 (comment)](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/issues/46#issuecomment-613487881)_
> 
> Eu vi que o manual do fork do @HPparanhos está bem legal.
> https://github.com/HPparanhos/CITI-OpenLung/blob/master/Manual%20de%20montagem/564135%20(1).pdf
> 
> Alguém teria uma sugestão a mais?

Otto, Boa tarde. Talvez minhas expectativas tnham sido errôneas. Andei vendo a parte burocrática de poder certificar o equipamento junto a ANVISA etc, e me parece que somente indústrias poderão ser habilitadas para a produção dos respiradores. Corrija-me, por favor, caso esteja errado. Minha expectativa era de poder fazer eu próprio (sou engenheiro mecânico e conheço uma coisinha ou outra), através de vaquinha ou outro processo de crow funding, a compra de peças e manufaturar algumas unidades para minha cidade, ajudar como fosse possível... Se obtivesse algum dinheiro de prefeituras próximas, talvez até produzir para essas pequenas cidades também. Moro em Tupã - SP, cuja população tem cerca de 60.000 habitantes. Não há indústrias capacitadas por aqui, só boa vontade mesmo (mas logo-logo teremos mortes). Por isso fiz a sugestão da visão explodida pra o equipamento, o que ajudaria (no velho sistema Lego) pessoas não técnicas a montar o respirador. Outra sugestão que faço, depois de todo esse preâmbulo, é um passo-a-passo de como inserir o firmware (aí talvez um entendedor de Arduino seja necessário). Um turorial, do início ao fim seria a perfeição para isso tudo, com locais explícitos pra o download dos softwares. Mas o tempo urge. Temos pouco tempo disponível. Talvez em duas semanas já estejamos iniciando o colapso do sistema público de saúde. Ponho-me à disposição para ajudar.

rafae2k, 2020-04-26:
@OttoHeringer Proponho a criação de uma animação (video) do processo de montagem, passo a passo, detalhando os pormenores do processo de montagem. Em um segundo momento podemos criar uma video aula acompanhando a instalação do firmware. Por último, podemos criar uma página nos moldes do [Ifixit.com](url) como este [https://pt.ifixit.com/Device/iPhone_11_Pro](url) ou nos moldes dos manuais da Prusa 3D [https://help.prusa3d.com/en/category/original-prusa-i3-mk3s-kit-assembly_280](url)

Cheguei agora na página do projeto e gostaria de saber se posso iniciar esta primeira ideia da criação da animação do processo de montagem ou se algo do gênero já está sendo realizado.