Inspire-Poli-USP/Inspire-OpenLung #50: readme.md Driver do motor DW322E

gdmpg, 2020-04-15:
# A que parte do projeto esse Issue se refere?

- [ ] Software
- [ ] Hardware
- [ ] Dados
- [X] Documentação
- [ ] Outro

### Reportando um erro

No setor de Partes Eletrônicas, consta 1x Driver DW322E da Leadshine após não encontra-lo verifiquei no side da empresa e encontrei o modelo DM322E  como pode ser visto em 

`http://www.leadshine.com/productdetail.aspx?type=products&category=stepper-products&producttype=stepper-drives&subtype=general-stepper-drives&series=dm&model=dm322e`

### Sugestão de melhoria

Verificar se realmente é uma versão diferente DW ou se é só um erro de escrita.

breno-helf, 2020-04-20:
Oi @gdmpg , muito obrigado pela Issue! Acredito que seja de fato uma versão diferente, mas acredito que o @OttoHeringer ou o @VitorFrost podem confirmar isso melhor.

Se for o caso você conseguiria fazer um Pull Request sugerindo essa mudança? 

Obrigado!

VitorFrost, 2020-04-27:
Infelizmente não sei responder, precisamos de alguém interno ao projeto para responder e confirmar.
MAS, no fork do CITI consta o modelo que mencionou!
https://github.com/HPparanhos/CITI-OpenLung

weverton-rosa, 2020-04-27:
Procurei também pelo modelo DW mas um dos fornecedores me retornou dizendo que desconhece essa versão e me orçou o modelo DM.

VitorFrost, 2020-05-01:
Pelos vídeos de divulgação, o driver já foi trocado por um outro de 24V. 
Pelas imagens, e por ser 24, deve ser o DM-542:
[image]
Fonte: https://www.youtube.com/watch?v=kMW-kKYsIrE