Inspire-Poli-USP/Inspire-OpenLung #51: Sensor de fluxo preciso e de baixo custo

drelp7, 2020-04-16:
# A que parte do projeto esse Issue se refere?

- [ ] Software
- [ X ] Hardware
- [ ] Dados
- [ ] Documentação
- [ ] Outro

Olá Pessoal. É possível desenvolver um sensor de fluxo de baixo custo e que consiga medir com resolução de 1 L/min. Para isso é necessário utilizar uma placa de orifício utilizada em baixas vazões e  pequenos diâmetros (norma ASME MFC-14M). Fiz algumas simulações teóricas aqui, e para conseguir uma resolução de 1 L/min é necessário ter um sensor que meça diferencial com resolução de 25 mmH2O (o MPX5010DP acredito que consiga isso). Digo isso, porque é importante que o respirador tenha o modo VCV no combate ao Covid, e para tanto é necessário medir e controlar o fluxo. No modelo por ambu, o fluxo pode ser facilmente controlado pela velocidade do motor durante a inspiração.

Fiz um protótipo aqui de placa de orifício, bem simples e funcional, que pode ser impresso em 3D. Assim que tiver os testes, caso se interessarem posso postar aqui.

Link com equações baseadas na norma ASME MFC-1M:  http://excelcalculations.blogspot.com/2011/10/small-bore-gas-orifice-meter-flow.html

Segue também em anexo arquivo do excel para fazer os cálculos.

[SmallDiaGasOrificeMeterASME2001-placa.xlsx](https://github.com/Inspire-Poli-USP/Inspire-OpenLung/files/4489780/SmallDiaGasOrificeMeterASME2001-placa.xlsx)

OttoHeringer, 2020-04-21:
Muito obrigado @drelp7 ! É exatamente esse tipo de contribuição que estamos procurando.

Você conseguiria fazer um pull-request dessa informações como uma página da wiki?

drelp7, 2020-04-21:
Boa tarde. Encomendei a placa de orifício por impressora 3D, acredito que chegue amanhã. Assim que eu montar aqui e tiver algum resultado posto pra vocês em um pull-request.

closed 2020-04-21 by drelp7

guiriole, 2020-04-21:
Boa tarde,
Desculpem, mas essa notícia procede?

https://www1.folha.uol.com.br/equilibrioesaude/2020/04/ventilador-pulmonar-da-usp-ja-pode-entrar-na-linha-de-producao.shtml

VitorFrost, 2020-04-23:
> 
> 
> Boa tarde,
> Desculpem, mas essa notícia procede?
> 
> https://www1.folha.uol.com.br/equilibrioesaude/2020/04/ventilador-pulmonar-da-usp-ja-pode-entrar-na-linha-de-producao.shtml

A sua pergunta é relevante, mas ficou fora de contexto. Verifique a informação e resposta nesse issue:
https://github.com/Inspire-Poli-USP/Inspire-OpenLung/issues/49#issuecomment-617166121

JonathasKerber, 2020-04-23:
Boa tarde. Qual foi a solução definida para o problema da medição do fluxo? Estamos com esse problema nos projetos aqui em Manaus.

OttoHeringer, 2020-04-24:
Oi @JonathasKerber. No projeto ainda não tem um sensor de fluxo, mas nos testes estamos usando esses equipamentos aqui, por exemplo:
https://github.com/Inspire-Poli-USP/Inspire-OpenLung/wiki/Sistema-de-Testes

drelp7, 2020-04-24:
Olá. Então, fiz alguns testes aqui usando uma placa de orificio em um tubo com 17mm de diâmetro interno e a placa com orificio de 6mm no centro e 3mm de espessura, utilizando um sensor de pressão diferencial com resolução de 1cmH2O a resolução vai melhorando de 5 Até 1L/min entre vazões de 10 e 60L/min. Vou fazer os testes utilizando um sensor de pressão com resolução de 0,20cmH2O. E com certeza a resolução da vazão vai melhorar bastante. Conseguindo 1L/min de resolução entre 10 e 60L/min é mais que suficiente.

JonathasKerber, 2020-04-24:
> Oi @JonathasKerber. No projeto ainda não tem um sensor de fluxo, mas nos testes estamos usando esses equipamentos aqui, por exemplo:
> https://github.com/Inspire-Poli-USP/Inspire-OpenLung/wiki/Sistema-de-Testes

Então vocês estão parametrizando o OpenLung através dos testes com essa instrumentação do link?  Como que fica a realimentação do sistema? Eu vi que vocês tiveram que fechar o algoritmo de controle, o que é uma pena, mas acontece. Eu sou de um instituto de P&D de Manaus, se houverem mais informações que possam ser compartilhadas, elas são bem-vindas.

Aqui estamos calibrando com o [Neós AVM100](http://www.neos.ind.br/produto/avm100/) e partindo para o diferencial de pressão para termos as medições de fluxo, é o que está parecendo ser o mais viável no momento.

OttoHeringer, 2020-04-24:
> Então vocês estão parametrizando o OpenLung através dos testes com essa instrumentação do link? Como que fica a realimentação do sistema? Eu vi que vocês tiveram que fechar o algoritmo de controle, [...]

A realimentação pode ser feita com esses equipamentos mesmo, que usamos principalmente para testar os algoritmos de controle.

Por favor, me procure no grupo do Telegram (https://t.me/openlungpoliusp) e me mande uma mensagem em privado para que eu possa orientá-lo sobre como solicitar acesso ao projeto completo. Desculpe não poder estar disponibilizando tudo aqui.

Luizhym, 2020-05-08:
Olá. Resolveram a questão do sensor de fluxo? Caso necessário, posso ajudar, tenho experiência no assunto.

drelp7, 2020-05-08:
Olá. Consegui desenvolver aqui um sensor de pressão diferencial, mas está dando um erro alto, de +-50 ml. Todavia percebi depois que a placa de orifício estava com vazamento. Mandei imprimir outra. Mas se quiser pode comentar a sua experiência que com certeza vai ajudar.

Luizhym, 2020-05-08:
Trabalhei com o desenvolvimento de máquinas de anestesia. Tive um contato com sensores de fluxo por diferencial de pressão, limitadamente, mas entendo em linhas gerais. Quais as dificuldades que estão enfrentando com este tipo de sensor? Poderia detalhar melhor?
Trabalhei bastante com sensores de fluxo do tipo fio quente, uma opção bem interessante.
Quais os requisitos para este sensor? Estão em alguma página ou documento?

drelp7, 2020-05-08:
Então se você quiser me add no whatss te explico melhor. N°: 44999529184.