Inspire-Poli-USP/Inspire-OpenLung #60: Questões jurídicas na utilização prática do equipamento na unidade de tratamento

adalmirabdala, 2020-04-23:
# A que parte do projeto esse Issue se refere?
- [x] Documentação
Considero necessária a explanação clara e objetiva da proposta do equipamento ao paciente (ou responsável), assim como o entendimento dos riscos e possibilidades de sucesso ou não do tramento. Daí vem a pergunta de como os centros de tratamento e hospitais destinarão esse recurso da forma responsável, clara e simultaneamente amparados pela legislação.

adalmirabdala, 2020-04-23:
Aqui coloco em questão que formulários, leis e considerações devem ser colocadas ao paciente ou familiar sobre o uso e riscos do equipamento. Eu, pessoalmente, encaro como um produto experimental a ser utilizado como último recurso, na escasses de outros, com o objetivo de prolongar a vida do paciente e ganhar tempo para a tomada de ações médicas.
É de se esperar que as instituições, por sua natureza de protocolos de atendimento bem regulamentados, necessitem de informações, certificações e amparo legal para a utilização de qualquer equipamento novo, assim como treinamento.
Já obtive comentários sobre o mesmo fato com medicamentos experimentais, onde existem regras claras e riscos claros, principalmente quando os recursos convencionais se esgotam e "não há mais o que fazer". Dessa forma acredito que todas essas questões sejam colocadas e discutidas para a utilização do equipamento de forma no mínimo com total ética e profissionalismo, para ter o aceite das instituições.

OttoHeringer, 2020-04-24:
O produto final, que usa parte do que está disponível nesse repositório, precisa ser validado antes pelos órgãos competentes.

Mas assumindo que a sua questão se refere ao escopo do bom uso pelo operador final, vou especular aqui com minha opinião pessoal (que não reflete a opinião da iniciativa): acho que não tem como evitar um uso inadequado disso, mesmo com tudo regulamentado e certinho. Mas acho que dá pra tomar duas medidas protetivas: um bom manual e fazer os hospitais/profissionais/pacientes assinarem um termo de consentimento sobre o uso do respirador emergencial.

Mas fico no aguardo de pessoas que têm mais conhecimento e experiência do que eu para fazer comentários mais precisos a respeito.

thauska, 2020-04-27:
https://g1.globo.com/sp/sao-paulo/noticia/2020/04/27/respirador-criado-na-usp-e-aprovado-em-testes-com-humanos-aparelho-e-feito-em-2-horas-e-15-vezes-mais-barato.ghtml

adalmirabdala, 2020-05-01:
Hoje obtive uma resposta de colegas da área de direito informando que em muitos hospitais já é procedimento pré-operatório de risco uma carta ao paciente/familia autorizando o uso de todos os recursos disponíveis, medicamentos e procedimentos em cáso de extrema necessidade ou falta de opções tradicionais. Neste caso apenas estenderíamos nesta mesma carta o assunto equipamento, de forma clara, respeitosa nas situações cabíveis, apresentando os riscos e benefícios. Assim que a informação mais refinada repasso aqui.