Inspire-Poli-USP/Inspire-OpenLung #66: Fuso Trapezoidal

andreuch, 2020-04-29:
# A que parte do projeto esse Issue se refere?

- [ ] Software
- [x] Hardware
- [ ] Dados
- [ ] Documentação
- [ ] Outro

   Estou com extrema dificuldade em encontrar um local para comprar o fuso, na especificação do projeto. Alguém pode informar onde posso comprar os seguintes componentes:
- Acoplador de Garra Drylin E para eixo redondo de 10 mm de diâmetro
- Fuso Trapezoidal de 10 mm de diâmetro, 15 cm de comprimento e passo de 25 mm
- Castanha de Latão TR10 e passo de 2,5 mm
   O passo e a castanhas são de 2,5mm ou 25mm?

eliaspereira9, 2020-05-01:
Tive as mesmas dificuldades na construção do projeto. O passo do fuso e da castanha devem ser idênticos para que possa funcionar perfeitamente. Este fuso de passo 25 mm não é vendido comercialmente, você terá que procurar algum profissional que usine o fuso e a castanha. Na minha cidade em Ponta Grossa - PR o valor mais baixo que consegui para a usinagem das duas peças foi R$ 600, sendo que caso alguém consiga mais barato aceito ajuda também pois pago o fuso e o frete.  Quanto ao acoplador você consegue encontrar em lojas de parafusos ou ferramentas mas antes precisa ter o diâmetro do rotor do motor pois varia entre modelos.

andreuch, 2020-05-01:
Ok. Agradeço as informações.
Atenciosamente,
André

Enviado do meu iPhone

bernardes01, 2020-05-13:
Olá,

Demorei para encontrar o fuso e a castanha, só achei quando coloquei o código que estava na lista de materiais no Google. Pelo que sei, só tem uma empresa que comercializa o fuso "stand", é a IGUS, fica em Jundiaí-SP, visitei eles semana retrasada.

Porem, há um problema, a IGUS importa os materiais da Alemanha, ou seja, caso acabe o estoque, demoraria para chegar a nova remessa, teríamos que achar um fornecedor secundário também, de preferencia que fabrique aqui no Brasil...