Inspire-Poli-USP/Inspire-OpenLung #67: Esclarecimento sobre questões jurídicas que impedem a prática da abertura

rpezzi, 2020-04-29:
Olá, 

Sou pesquisador e não conheço restrição jurídica que nos impedem de praticar o desenvolvimento de software livre ou praticar a ciência aberta. Por isso fiquei um tanto preocupado com a decisão do Inspire não utilizar modelos abertos para o desenvolvimento do software de controle, assim como não praticar a ciência aberta para o ensaios, alegando questões jurídicas. A USP é referência internacional e uma mudança de rumo de um projeto que se lançou e continua se promovendo como aberto merece esclarecimento. 

Sendo o firmware uma parte essencial de um respirador, que garantias aqueles que estão reproduzindo e modificando o hardware do projeto poderão utilizar o firmware em seus equipamentos? Quando terão acesso ao firmware? Poderão fazer ajustes no firmware para utilizá-los em versões modificadas do Inspire? Essas são questões que evidenciam como a decisão de fechamento de uma parte do projeto afeta o impacto e alcance de toda a iniciativa. É natural que o projeto possa necessitar de ajustes de acordo com a disponibilidade de peças, portanto firmware abeto é essencial.  
 
Este é um momento crítico da humanidade onde as melhores ideias devem se espalhar o mais rapidamente possível. Entendo que essa foi uma das motivações de lançar abertamente o Inspire. Assim, ao mesmo tempo que pergunto quais são as questões jurídicas - se é algo interno da USP, ou algo que se aplica no Brasil todo? -,  gostaria de sugerir que essa decisão seja reavaliada em prol do bem comum.

Att.
Rafael

emersonmoretto, 2020-04-29:
Olá Rafael,
sou um dos desenvolvedores do firmware do projeto, sou à favor de códigos
abertos também.

Porém é necessário ponderar alguns pontos, todos pontos abaixo são minhas
opiniões pessoais e não representa necessariamente as opiniões dos
coordenadores do projeto.

1) Há uma questão jurídica apontada pelo departamento de direito que nos
apoia na qual foi destacado que o um equipamento médico e/ou com fins
terapeuticos fora da conformidade implica em* crime contra a saúde pública *
(
https://www.invitare.com.br/arq/legislacao/anvisa/Manual-para-regulariza-o-de-equipamentos-m-dicos-na-Anvisa-2010-ANVISA.pdf
)

Confesso que não sou um especialista em leis, porém aqui vai um
destaque em *uma
das leis que implica *nisso (parece que é uma composição de várias leis):

*Pena - reclusão, de 10 (dez) a 15 (quinze) anos, e multa.*         (Redação
dada pela Lei nº 9.677, de 2.7.1998)
<http://www.planalto.gov.br/ccivil_03/LEIS/L9677.htm#art273>

§ 1º-B - Está sujeito às penas deste artigo quem pratica as ações previstas
no § 1º em relação a produtos em qualquer das seguintes condições:
    (Incluído
pela Lei nº 9.677, de 2.7.1998)
<http://www.planalto.gov.br/ccivil_03/LEIS/L9677.htm#art273>

I - sem registro, quando exigível, no órgão de vigilância sanitária
competente;         (Incluído pela Lei nº 9.677, de 2.7.1998)
<http://www.planalto.gov.br/ccivil_03/LEIS/L9677.htm#art273>

II - em desacordo com a fórmula constante do registro previsto no inciso
anterior;         (Incluído pela Lei nº 9.677, de 2.7.1998)
<http://www.planalto.gov.br/ccivil_03/LEIS/L9677.htm#art273>

*III - sem as características de identidade e qualidade admitidas para a
sua comercialização;*         (Incluído pela Lei nº 9.677, de 2.7.1998)
<http://www.planalto.gov.br/ccivil_03/LEIS/L9677.htm#art273>

2) Além de toda essa questão jurídica, temos que ter muita
responsabilidade. O firmware ainda está em desenvolvimento, agora estamos
na parte de detecção de alertas de intercorrências. O ventilador tem que
emitir alertas (seguindo as normas ABNT disponibilizadas na RDC que nao
lembro o número). Isso é uma parte MUITO crítica do sistema. É justamente
onde o paciente possa ter um imprevisto sério e o sistema tem que detectar
isso.

3) Lembrando que a decisão de liberar esse código fonte precocemente é dos
coordenadores do projeto, cabe a eles toda a responsabilidade legal também
né. E, de acordo com eles, isso será liberado no momento certo, quando
estiver mais maduro e aplicável, com menos probabilidade de causar algum
dano sério a algum paciente.

Particularmente, penso que é muito perigoso soltar um código ainda não
totalmente pronto/testado de forma aberta. A gente não consegue garantir
que alguém possa reproduzir isso com segurança e sair testando (com aval de
algum comitê de ética em pesquisa) em pacientes e animais. Qualquer
imprevisto desse pode, além de causar algum dano muito sério a algum
paciente, pode também por consequência causar o fim do projeto. Além disso,
as questões legais se aplicam diretamente aos autores do projeto.

Por fim, toda a equipe do projeto é composta por voluntários, que estão
trabalhando há +- 40 dias de domingo a domingo, pelo menos 12h por dia. Nós
estamos MUITO preocupados com o momento crítico que o mundo está passando e
estamos procurando fazer nosso melhor e com muita responsabilidade.

*Emerson Moretto*
emerson@moretto.eng.br
11 99484-0571

solstag, 2020-04-29:
Oi Emerson,

Eu já trabalhei com software livre em ambiente hospitalar. Apesar do meu software não intervir diretamente em pacientes, ele era parte do processo de diagnóstico.

Vou explicar aqui por quê do meu ponto e vista vocês estão sofrendo e promovendo FUD ([FUD](https://fr.wikipedia.org/wiki/Fear,_uncertainty_and_doubt)).

Primeiro, o artigo da lei trata de produtos destinados ao consumo, distribuição e comercialização. Não tem como enquadrar a publicação de um design ou instruções de computador para fins de desenvolvimento nisso, pois não se trata de produto a ser distribuido, nem é destinado ao consumo. Seria como dizer que é um crime as publicações de protocolos farmacêuticos, sem as quais a pesquisa em farmacologia sequer poderia existir, alegando que alguém poderia produzir medicamento em casa.

Segundo, que nas cicunstâncias atuais, de emergência sanitária, e partindo de uma instituição séria, com contatos no governo e no mundo jurídico, não há o menor risco real de um processo. O direito não se exerce num vácuo, e me soa muito falso alguém sugerir que a justiça irá atrás de algum professor da Poli por acelerar e promover colaboração ao divulgar conhecimento crítico para combater a epidemia.

Assim, a interpretação que você apresenta não corresponde nem à letra nem ao espírito da lei.

Terceiro, se não se tratasse de FUD, existiria a opção de divulgar de forma controlada, num repositório privado, para outros grupos interessados. Me parece que nem isso está sendo feito. Se o projeto é tocado por voluntários, e se vocês consideram de fato Open Source como uma forma eficaz de tocá-los, o que faz alguns voluntários melhores que outros? As lições mais básicas do Open Source nós sabemos: os bugs ficam rasos e o desenvolvimento acelera. Com a quantidade de gente capacitada interessada, você tem alguma dúvida que esse software seria mais seguro e ficaria pronto antes se estivesse de fato sendo desenvolvido abertamente?

Ao meu ver, o que está sendo feito é atrasar a depuragem e melhoria desse firmware, desviando energia para uma discussão que não deverai estar sendo uma barreira para mais pessoas contribuirem e avançarem o projeto. E, lamentavelmente, provavelmente custando vidas.

E enquanto isso ...

https://www1.folha.uol.com.br/cotidiano/2020/04/doria-compra-respiradores-sem-licitacao-da-china-por-r-550-mi-promotoria-investiga.shtml

Um abraço,
.~´

rpezzi, 2020-04-29:
Olá Emerson, 

Obrigado pela pronta resposta e parabéns pelo esforço e dedicação para todos envolvidos. Minha sugestão de reavaliar a abertura do projeto visa justamente amplificar o impacto do esforço de vocês. Afinal, um respirador barato e relativamente fácil de montar pode salvar muitas vidas. Creio que manter os princípios de abertura seja a melhor forma de fazer isso.

Me preocupa também pois caso o firmware ou outras partes do projeto não sejam disponibilizadas abertamente, creio que este será o primeiro caso no Brasil de uma grande instituição promover um projeto como aberto, utilizar o marketing do open, lançando financiamento coletivo, chamada pública para contribuir, para depois fechar. Enfim, abre precedente que enfraquece o movimento open source.

Sobre categorizar a abertura do projeto como crime contra a saúde pública, creio que seja um exagero. Comercializar e aplicar algo fora de especificação e registro sim, pode ser crime; Mostrar o andamento do desenvolvimento não. Se a publicação de um projeto de equipamento médico fora de conformidade fosse crime, o que já está publicado neste repositório já se enquadraria - o firmware não muda isso. Também seriam crimes contra a saúde pública a publicação de livros didáticos com projetos de instrumentos de diagnósticos, tais como circuito de eletrocardiograma , ou bibliotecas ou esboço de software para análise de imagens médicas... 

Sobre as preocupações com mau uso que acarretem riscos, o risco existe independente da data de lançamento do firmware. Com ou sem o firmware desenvolvido por vocês, os riscos existem. Também, independente do firmware: nunca conseguirão, nem é o seu papel, "garantir que alguém possa reproduzir isso com segurança e sair testando (com aval de algum comitê de ética em pesquisa) em pacientes e animais".  Esconder partes do projeto não ajuda aqui. Os avisos já presentes na documentação deixam claro que os equipamentos não estão em conformidade, não devem ser comercializados para uso em pessoas nem em animais sem as aprovações das instâncias competentes. 

Rafael

emersonmoretto, 2020-04-29:
Bom, vou tentar responder ambos rapidamente, novamente com minhas opiniões pessoais.

> Segundo, que nas cicunstâncias atuais, de emergência sanitária, e partindo de uma instituição séria, com contatos no governo e no mundo jurídico, não há o menor risco real de um processo. O direito [...]

> Assim, a interpretação que você apresenta não corresponde nem à letra nem ao espírito da lei.

Eu acho que isso "não há o menor risco real de um processo" é a sua interpretação e seu ponto de vista, o que não garante que um juiz possa ter a mesma interpretação no caso de um aparelho desses tirar vida de pessoas. E, novamente, essa decisão cabe aos coordenadores do projeto. Eu, particularmente, concordo em partes em segurar um pouco o código fonte enquanto ele não estiver minimamente seguro. É MUITA responsabilidade nisso. É um projeto que tem muita visibilidade por ser da USP, por estar em fase avançada de testes.. O projeto MIT e-vent me parece que também não soltou o código fonte do firmware (não tenho certeza disso). Tem muita gente movida pelo desespero da situação e **como já aconteceu** de sair reproduzindo o projeto e colocando em testes com animais.

> Sobre categorizar a abertura do projeto como crime contra a saúde pública, creio que seja um exagero.

Novamente creio que isso não depende exatamente do que pensamos mas sim, do juiz que viria a julgar um caso desses. E talvez não seja exatamente um exagero, pois será preciso uns 40 mil ventiladores emergenciais desse tipo e, infelizmente, há um risco disso poder causar danos e até óbitos em escala considerável.

>  creio que este será o primeiro caso no Brasil de uma grande instituição promover um projeto como aberto, utilizar o marketing do open, lançando financiamento coletivo, chamada pública para contribuir, para depois fechar. 

Isso não está sendo usado para promoção. É muito negativa essa insinuação e não é construtiva. É uma responsabilidade enorme esse projeto, e uma exposição pessoal (meu número de whatsapp está no rodapé e durante as primeiras semanas do projeto tava na página principal no github). Eu poderia tranquilamente estar em casa durante esses 40 dias, fazendo meu doutorado, andando com minhas pesquisas aqui da usp. Além disso, Inicialmente o projeto era integralmente open source, inclusive o primeiro firmware. Após recebemos essas recomendações jurídicas, foi fechado o código do firmware pois poderia caracterizar que estaríamos oferecendo abertamente um equipamento médico.

> Os avisos já presentes na documentação deixam claro que os equipamentos não estão em conformidade,

De acordo com as questões jurídicas não, pois a gente teria que garantir que no momento que a pessoa pegou nosso código ela tenha ciência disso através de algum termo oficializando isso.

>. Se o projeto é tocado por voluntários, e se vocês consideram de fato Open Source como uma forma eficaz de tocá-los, 

A gente se considera open source, tanto é que no começo era. Hoje mantemos aberto o que teoricamente não oferece riscos jurídicos e à saúde de animais e pacientes e, espero que, muito em breve seja disponibilizado o firmware.

> o que faz alguns voluntários melhores que outros? 

A gente não se considera melhores do que outros. Todos voluntários aqui são da usp, são pesquisadores daqui da poli que toparam o desafio de vir aqui trabalhar fisicamente no projeto (pois, infelizmente, praticamente todos os testes precisam ser feitos no pulmão artificial aqui do laboratório). Caso consiga participar presencialmente aqui (infelizmente por conta do pulmao artificial), posso ver com os coordenadores de vc fazer parte do time.

Por fim, acredito que se a comunidade estiver bem determinada, seria interessante uma iniciativa de desenvolver um código aberto em paralelo. Obviamente não é o jeito mais produtivo, porém é mais uma chance de dar certo, é mais gente tentando e as opiniões de riscos e jurídicas ficam a critério de cada autor.

Bom, vou voltar pro código aqui.

jonasmalacofilho, 2020-04-29:
@emersonmoretto 

É importante considerar a distinção entre equipamento, especificação e código-fonte.  Me parece que isso _pode_ ter sido perdido na comunicação com o departamento jurídico.  

Uma linha de investigação específica pode ser se, dentro do limite da razoabilidade, os dois últimos se enquadraram no Art. 5° da Constituição, informalmente: "liberdade de expressão", explicitamente mencionadas atividades intelectuais e científicas.

Digo "dentro do limite da razoabilidade" porque o argumento de "que não garante que um juiz possa ter a mesma interpretação" é falho: o juiz pode ter a tal outra interpretação (de que vocês são responsáveis) mesmo que que o código-fonte seja completamente fechado.

Mas volte para o código, e parabéns/obrigado pelo trabalho... apenas veja se essa distinção não ajuda a dar mais segurança jurídica para a abertura completa do projeto.

Jonas
(POLI/mecânica, 2011)

solstag, 2020-04-30:
Ni! Oi Emerson,

Eu entendo sua preocupação e só tenho o que admirar nas suas motivações.

Isso obviamente não é problema seu, aliás eu não entendo porque você está respondendo aqui se você não é o responsável por essa decisão.

Agora, na boa, o depto jurídico da USP fez FUD pra cima de vocês e vocês compraram. Mas na atuação deles, são essencialmente pagos pra fazer FUD - e nem sempre num mal sentido, mas é o hábito deles, além de não terem preparação para lidar com a realidade que estamos vivendo.

Os professores tem bem consciência desse hábito. Alguns dos projetos mais bacanas em que trabalhei, se dependessem do jurídico da USP não teriam jamais saído do papel.

Como eu tentei explicar, essa interpretação aí não se sustenta. E não depende de juiz, certamente não mais do que um juiz poderia já processar vocês e todo mundo fazendo projetos de ventilador - e teria com isso que justificar por que não processar também todo mundo que já publicou uma receita de fármaco ou patenteou um projeto de dispositivo médico.

Outro engano é essa ideia de que estão evitando algum desastre ao adiar a publicação do código. O mundo está a par de 3n+1 projetos DIY de ventiladores, tem listas com mais de 200. Quem for testar de forma irresponsável vai fazê-lo com ou sem vocês. Se o código de vocês é melhor e não está público, só estão garantindo que as pessoas usarão firmwares menos próprios, e que a experiência adquirida não vai contribuir para a melhoria do de vocês.

Espero que possam trazer o Inspire em forma open-source o quanto antes. Eu não tenho veículo pessoal e estou longe do Butantã, mas agitei uma galera com quem tenho trocado ideias desde o começo de março. Pelo menos um esteve com vocês aí na Poli contribuindo, outros contribuiram documentação remotamente. Certamente há outros grupos. Nós estamos aí, caso tenham tarefas a distribuir, que não exijam o equipamento, ou mesmo que exijam, pois alguns estão em instituições onde potencialmente teriam acesso a pulmões artificiais.

No mais, bom trabalho e toda força pra vocês!!

.~´

emersonmoretto, 2020-04-30:
Valeu pelo apoio pessoal!!!

>  eu não entendo porque você está respondendo aqui se você não é o responsável por essa decisão.

É... Só tentei responder por ser um assunto bastante pertinente e, de certa forma, tentar explicar a responsabilidade que o projeto tem e de seus responsáveis legais.

Em tempo, o firmware em si não está atrasando o desenvolvimento do projeto, tem toda a parte de homologação, ensaios físicos/mecânicos/elétricos, tem a parte de manufatura, compras, cadeia de fornecedores, importação, etc etc etc.. tudo isso andando em paralelo. O código fonte está em fase de aprimoramento e testes massivos em pulmão artificial, enquanto o resto do projeto anda em paralelo. Estou bastante animado com os resultados e realmente tou muito esperançoso que isso possa ser aberto e possa ser fabricado muito em breve.

Obrigado pela compreensão e/ou respeito sobre esse assunto e, de fato, é preciso concordar que é um assunto muito delicado.

obrigado!!!

VitorFrost, 2020-04-30:
Aos responsáveis pela não liberação do projeto,

Espero que quando liberarem o código não tenham falhas ocultas que teriam
sido sanadas, caso o código já estivesse aberto. Afinal, nessa posição,
estão assumindo que não haverão erros.

Se é pra usar opinião e não embasamento jurídico:
Na minha opinião, estão indiretamente promovendo o uso de software não
oficial, que não tem garantias! Estão assumindo o risco de alguém replicar
o projeto sem um código seguro!
É um risco por omissão, como o risco de alguém que atropela e não assiste!

 Se você está querendo se resguardar, crie um termo de uso e ou acesso o
quanto antes, e pronto!

emersonmoretto, 2020-04-30:
> Espero que quando liberarem o código não tenham falhas ocultas que teriam sido sanadas, caso o código já estivesse aberto. Afinal, nessa posição,
estão assumindo que não haverão erros.

Mais uma vez: o código está fechado por enquanto por questões de interpretação e responsabilidade jurídicas e não porque estamos prometendo um código perfeito sem erros.

É um direito de quem está assumindo a responsabilidade legal sobre isso. É fácil julgar que é um exagero quando a responsabilidade legal disso não é contigo. 

Se algum de vocês realmente quiser ajudar no código fonte, meu WhatsApp tá aí, posso ver a autorização pra entrar na usp e vir aqui no laboratório pra trabalhar conosco nos testes com o pulmão artificial.

eduoda, 2020-04-30:
Condicionar a ajuda ao trabalho presencial não é lá muito razoável, mas para além disso, existem outros grupos geograficamente distantes que poderiam imediatamente se beneficiar do compartilhamento e vice-versa.

No mais, se a questão é legal, podemos trazer para o debate pessoas com experiência (legal) no assunto, boa parte de quem comentou aqui está ligado a organizações, nacionais e internacionais, dentro e fora da academia, que tem esse foco.

Olhando de fora, a gente tem que se esforçar para não entender que a questão jurídica tem sido uma desculpa para outras motivações para fechar o projeto, não me assusta se daqui a pouco aparecer um papo de patente.

adalmirabdala, 2020-05-02:
Ponderando o que entendi:
1) o que suponho ser negativo para USP:
- abrir o código num momento considerado "tudo OK" chama a responsabilidade de um openSource 100% para USP. Realmente abrir para o pessoal testar, muda o formato da segurança do projeto e divide a responsabilidade.
- o projeto iniciando openSource e fechando na mão de uma empresa detentora dessa tecnologia colocaria em XEQUE a UTILIZAÇÃO dos RECURSOS PÚBLICOS (USP), não importando que questão júrídica que estão justificando o fechamento do projeto (que é outra questão diferente).
- poucos tem acesso ao conceito das curvas biológicas, o que NÃO É O CÓDIGO e poderia ser ABERTO. Se tudo for aberto de uma vez será muita informação par se aprender de uma vez só. Haverá problemas maiores. O melhor jeito de ninguém ficar fazendo experiências perigosas é acessar os gráficos e formulas prontas, acessos biológicos que só a USP tem recursos.
- testes e verificação de bugs mais demorados (problemas que existem em todos os projetos novos no mundo).

2) o que seria positivo no projeto fechado:
- confiança de replicação do projeto e garantia de não alteração de pontos testados.
Bom. Isso seria resolvido facilmente com um contrato de uso colaborativo responsável a um grupo que assuma os termos de compromisso.

O que acredito estar deixando todos os co-projetistas ansiosos é NÃO TER até o momento nenhum pronuncioamento de que será aberto algum dia, o que é PREOCUPANTE MESMO !!
E nenhuma política de distribuição ou critério objetivo para ter acesso ao projeto foi divulgado, o que no Brasil leva a interpretação de apadrinhamento (o que acredito não ser verdade até agora).

Enfim, eu acabo ficando de ruim na história ao levantar estas questões, mais se estas políticas não forem colocadas de forma CORRETA e clara LOGO, é o que vai aparecer no jornal. 

Fato é que o projeto está aí por todo o lugar com a cara da USP, já iniciada pela USP e se alguém fizer um projeto falho a USP já é responsável juridicamente. Vários pontos desde o lançamento da idéia até os materiais que tornam possível a realização de um protótipo funcional estão por tudo.
Por esse ponto de vista, o melhor seria novamente a opção de apoiar, ensinar (o que não está acontecendo, sem precisar abrir o código) e distribuir para quem se comprometer em auxiliar.

Tentei concatenar alguns pontos positivos/negativos.
Enquanto isso os hopitais estão lotando.
Assumir riscos faz parte do compromisso da posição e se há impeditivos para executar, seria prudente algum coordenador ser mais transparente declarando necessidades e as fases.

Vejo o Otto e o Emerson só segurando bomba, que na minha opnião dificulta a coletar opiniões voltadas a resolver os problemas técnicos, tirando-os do foco de finalização.

Pergunta... para que lado estamos caminhando ???

VitorFrost, 2020-05-02:
Complementando, nós temos duas maneiras de enxergar a situação:
Primeira - Vocês fecharam o projeto por puro orgulho e estão utilizando o medo legal como justificativa.
Segundo - Vocês realmente estão com medo, e a parte da assistência jurídica esta admitindo a solução mais fácil, que é empurrar com a barriga.

Supondo que seja o segundo caso, eu afirmo que vocês estão sendo mal assessorados juridicamente!
Vocês estão se matando por dias pra conseguir um respirador de baixo custo funcional. Vocês sujaram as mãos ao invés de ficar na posição mais confortável que é: esperar que os governos comprem os produtos dos fabricantes certificados.
O que não está acontecendo é o mesmo esforço da parte jurídica, dizendo que é possível manter o projeto aberto e vocês protegidos, eles estão no conforto não apresentando soluções.
Enquanto isso, vocês ficam nesse sanduíche entre as pessoas cobrando o projeto aberto e o jurídico pondo medo do outro lado! Quando deveria ser o contrário, a barreira jurídica entre vocês e a comunidade!
Quer ver como não faz sentido? Vocês estão utilizando um código aberto, o Debian, na plataforma labrador. Se o equipamento final apresentar problemas, não faz sentido culpar a Debian por uma falha do sistema operacional, se o ventilador falhar!
A pergunta não é se é possível deixar aberto ou não, mas como fazer isso garantindo a segurança jurídica dos envolvidos no desenvolvimento!
E quanto mais aberto o projeto, mais diluída vai ficar a responsabilidade por erros!

felipesanches, 2020-05-04:
Não se trata exatamente do mesmo problema (não é equipamento médico), mas há outro caso relativamente recente **vindo do mesmo laboratório da Poli-USP** em que "questões legais" foram apresentadas como motivo para um projeto divulgado como se fosse hardware/software livre, não ter publicado ainda os arquivos de projeto do hardware: https://github.com/caninos-loucos/CaninosSDK/issues/1

rpezzi, 2020-05-17:
Compartilho aqui [este post do Forum GOSH](https://forum.openhardware.science/t/covid19-open-source-hardware-legal-recommendations-for-makers/2300) que contém alguns links com orientações legais para makers atuando em ações contra o COVID-19.

Tem várias referencias relevantes para a equipe que está dando orientação jurídica ao Inspire. Particularmente relevante é a questão da responsabilidade e as exceções. Dentre os links apontados, destaco a [Seção 4.2.1 deste documento](http://www.didiy.eu/public/deliverables/didiy-d6.1.pdf), que trata das exceções das responsabilidades de quem produz os dispositivos médicos na legislação europeia.  

É urgente uma análise frente à legislação brasileira para termos certeza se os crimes contra à saúde pública citados acima podem ser atribuídos para quem pesquisa e produz material educacional sobre esses equipamentos médicos. 

Enquanto existem diversas comunidades de tecnologias livres se organizando ao redor do mundo para combater a pandemia, bem ilustrado pela [Vent-con 2020 - An Open Source Ventilator Conference](https://www.pubinv.org/ventcon/), o Inspire muda suas orientações com uma justificativa que impacta toda comunidade, pois explicitamente passa a mensagem de que é arriscado contribuir para projetos open source de instrumentos médicos aqui no Brasil.
 
Por enquanto temos relatos de orientações informais do departamento de direito da USP que sugere que pesquisadores podem ser responsabilizados pelos atos de terceiros, o que está prevenindo a disseminação de tecnologia que pode salvar muitas vidas. Dada a urgência imediata para enfrentar a pandemia, e a necessidade de esclarecer a segurança jurídica de projetos open source de uso médico no Brasil, seria adequado à USP fazer um esclarecimento oficial.

felipesanches, 2021-08-31:
Nesse ritmo acho que superaremos a crise de saúde pública antes de conseguirmos superar essa questão política com "roupagem de juridiquês"... A quem poderia interessar essa morosidade toda? O que podemos fazer para desenroscar esse nó?

enieber, 2021-09-01:
qual o contato da parte juridica? Quem pode responder por eles para a criação da licensa de software que permite a abertura do codigo sem a responsabilização dos criadores e pesquisadores do projeto?

enieber, 2021-09-01:
Até onde vi, ja tem uma licensa, ja tem alerta e responsabilização. Qual o professor e quem pode conversar publicamente nessa issue para a discução da desprivatisação  o CODIGO PUBLICO?

enieber, 2021-09-01:
Esperando o posicionamento juridico da USP ou dos cordenadores para esclarecimento.

irismar, 2022-08-29:
A questao é simples quando o projeto estava em andamento era aberto varias pessoas ajudaram volutariamente a escrever codigo e propor solucoes depois do projeto pronto fecham o projeto sob alegacao ridicula se derem a receita do bolo sao responsaveis legais pelo seu  preparo e gosto. a questao que querem vender o codigo e usar a maquina como encubadora do codigo